FreeCAD/FreeCAD #18406: Assembly not solving even on small number of parts/joints

Shkolik, 2024-12-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Issue is kinda similar to #17917, but not much parts involved and I wasn't able to find workaround. 
Video below shows my test assembly using Fixed joints:

<video src="https://github.com/user-attachments/assets/a4f64fed-1c9c-42fe-b2f2-4a244bd806d1">Fixed joints</video>

Parts are fixed where they should be, but when trying to move what suppose to move - bunch of exceptions and nothing moves. 

Here is my attempt to use distance joints instead of fixed:

<video src="https://github.com/user-attachments/assets/bbab8f59-fe4a-43f1-88d1-7efe92fbcfa3">Distance joints</video>

Parts are everywhere, warnings and errors in the report view.

Test files in attached archive:
[assy_tests.zip](https://github.com/user-attachments/files/18071702/assy_tests.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CurvedShapes 1.0.8
  * Curves 0.6.28
  * fasteners 0.5.21
  * Foamcut 0.1.5
  * Marz 0.1.5
  * OpenTheme 2024.9.1
  * PieMenu 1.8.1
  * SmartReportFields1.xlsx
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-12-13:
Your assembly is overconstrained. You have 2 sliders when you need only 1.

Shkolik, 2024-12-14:
> Your assembly is overconstrained. You have 2 sliders when you need only 1.

Yes, already realized it, BUT! user should not see billion of errors in a console - it should be only one saying that there is overconstrained condition. I tried to reproduce it again with only one slider joint - error is gone, but billion warnings still there. See attached video:

<video src="https://github.com/user-attachments/assets/b6d09b36-9186-480c-a28a-56926359a046">one slider</video>

At least it's not failing apart now.

PaddleStroke, 2024-12-16:
You are correct. We need to have a way to detect overconstraints and feedback to the user correctly. Perhaps you could modify this issue report to this? "Assembly not reporting overconstraints" or something. Though there's maybe already an issue for this.

In your latest video I see you have the 'bad solve ignored' warnings. I'm not sure why it would give you this warning. Can you give me the file so I can test?
Besides maybe bad solves can be ignored silently, there's not much value in showing this message thousands of time anyway.

Shkolik, 2024-12-16:
File is basically the same as attached above, but here it is:
[assy_test.zip](https://github.com/user-attachments/files/18151963/assy_test.zip)

kozross, 2025-01-09:
I've run into a _very_ similar problem before, and would have loved for the solver to have been more informative. Ideally, I would like the solver for Assembly to be at least as informative as the one for Part Design when it comes to overconstraints: firstly, it should report them (instead of silently breaking as it does now), and it should specify which joint(s) are forcing the overconstraint.

PaddleStroke, 2025-07-23:
I think I will tackle this issue next. We need something similar to sketcher to say how many DoF we have. Probably with the same widget as Sketcher.

PaddleStroke, 2025-07-24:
So the plan is this : 

[image]

Now the main issue is how to manage this taskbox. Because it's not as straight forward as it seem.
- We want to show this taskbox when the assembly is active.
- We want to still be able to start and use tools. The issue is that the taskview is used by the tools. Say you start the 'insert component' tool. What it does is : 
```
def isAssemblyCommandActive():
    return activeAssembly() is not None and not Gui.Control.activeDialog()
```
So the tool is only active if there is currently no activeDialog. ie if there is no taskbox. Because that usually mean that another tool is executing.

Long story short, using the existing way to show a taskbox would conflict with tools that need a taskbox. So that's not possible.

The other possibility is something like TaskWatcher of PartDesign. It's a taskbox that is tied to the workbench. However the way it's designed is not good either : It's tied to a workbench and not an edit mode. And it's designed to show commands.

So I think we need to define a new framework. That would be for 'permanent edit taskbox'. This way as long as the assembly is in edit we show this.

By the way a TaskWatcher could also be useful for user to discover the workflow. It could show things like 'Activate assembly' 'insert part' 'ground part'...

PaddleStroke, 2025-07-25:
[image]

That work. Now on to the actual DoF calculation etc.