FreeCAD/FreeCAD #18409: Ubuntu 24.04: No miniature images on FreeCad start page

Gerd-St, 2024-12-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am using FreeCad (snap version) on a Ubuntu 24.04 and when the FreeCad start page gets displayed it shows only empty sheet icons for both examples and my recently opened projects. The settings for including miniature pics and FreeCad logo are active.
This problem ranges back at least for 6 months with various FreeCad dev builds and also occurs with official version 1.0.

I additionally installed already libqt5svg5-dev, libqt5x11extras5-dev, qttools5-dev and qttools5-dev-tools but without success.
What is missing?

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1202
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-10:
Please test on the latest weekly, should already be resolved.

Gerd-St, 2024-12-10:
Just updated to the dev version, problem still exists:
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39562 (Git) Snap 1221
Build type: Release
Branch: main
Hash: 16ac5f99770e078f97e07d7bcf06654f90846673
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

furgo16, 2024-12-10:
@Gerd-St thanks for the report. This is a duplicate of a known issue: https://github.com/FreeCAD/FreeCAD/issues/13676. Please feel free to report any new findings there.

closed 2024-12-11 by luzpaz