FreeCAD/FreeCAD #18413: creation of line from midpoint

Roberto8303, 2024-12-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
[image]

Is it possible to implement the function of creating a line starting from the midpoint like in other software?

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.13
  * ExplodedAssembly
  * fasteners 0.5.33
  * FeedsAndSpeeds 0.5.0
  * FreeCAD-themes 2024.7.24
  * Glass
  * IconThemes
  * OpenTheme 2024.9.1
  * PieMenu 1.10.1
  * Render 2024.11.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-10:
This possibility already exists

https://github.com/user-attachments/assets/d7f59f97-8556-40bd-82cc-47b1d7853078

Roberto8303, 2024-12-10:
@tarman3 not this one. 
I intend to create a line that starting from a point extends equally to the right and left

Roberto8303, 2024-12-10:
https://www.youtube.com/shorts/VVH6kEgTfxM?feature=share

Roberto8303, 2024-12-11:
@yorikvanhavre can it be done?

Pendelf, 2025-03-22:
It's not a bad idea, but it's rare in CAD

RustoMCSpit, 2026-02-15:
also, the slot tool should have this feature as well.