FreeCAD/FreeCAD #18416: TechDraw Broken View, grey background of the region of the broken view

Norbert-Rotorwerk, 2024-12-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In TechDraw Broken View the background of the broken region has a grey background after exporting the drawing to pdf and or svg format. The background colour is f1f3f5ff in RGB (R 241; G 243; B 245). The colour should be white (RGB ffffffff). 

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (MATE/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * BOLTSFC 2022.11.5
  * parts_library.bak
  * parts_library
  * symbols_library
  * Defeaturing 1.2.0
  * Assembly4 0.12.6
  * Manipulator 1.5.0
  * MeshRemodel 1.8919.0
  * Render 2023.2.2
  * Help 1.0.3
  * FEM_FrontISTR 0.1.0
  * Assembly3 0.12.2
  * QuickMeasure 2022.10.28
  * fasteners 0.5.29
  * 3DfindIT 1.2.0
  * sheetmetal 0.4.26
  * BillOfMaterials 1.0.1
  * Design456 0.0.1
```

### Subproject(s) affected?

TechDraw

### Anything else?

[Broken-View-grey-region.pdf](https://github.com/user-attachments/files/18082702/Broken-View-grey-region.pdf)
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-10:
@Norbert-Rotorwerk can you zip up the .FCStd file and drag-drop it into Github textbox so we can test it conveniently ?

Norbert-Rotorwerk, 2024-12-11:
I got it. The page colour is not white. It is grey. But it must be the default colour after installation of FreeCAD. I changed it to white and now it is okay.  
Thanks WandererFan!

closed 2024-12-11 by luzpaz

WandererFan, 2025-02-05:
@Xico-DE - can you share your FCStd file?

Xico-DE, 2025-02-06:
Hi Norbert,
Thank you very much for the help. 
It did work!