FreeCAD/FreeCAD #18417: TestPathToolController test fails if user settings have insufficient precision

kpemartin, 2024-12-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If the user Preferences have insufficient precision, the `TestPathToolController` tests fail because the tool diameter of 1.75mm does not make a round-trip accurately enough.

In particular, if the user has selected Imperial decimal units (inches, etc) but not changed the default of 3 decimal places, the test will run with these settings and fail because the round trip passes through the string representation "0.069 in" (perhaps using some other representation for the 'inches' unit), which then converts back to 1.7526mm. This is not within the 0.00001 error allowed by `assertRoughly` so the test fails.

The "0.069" is the 3-digit rounding of the more accurate value of 0.0688976377952755905511811023622 inches.

Conversely, the test also uses `assertRoughly` for other checks even though there is no reason the values should not be exactly equal.

The test should set the precision and units sufficiently to assure the round-trip value is accurate, or use a more lenient error threshold based on the units and precision. If the test changes the settings it must restore them on exit.

It is not clear how the test can effectively change these settings. Although it could (probably) reach in and alter the Parameters in `User parameter:BaseApp/Preferences/Units/Decimals` and `.../UserSchema` it does not appear, from reading the code, that this will have an immediate effect on calculations. These parameters are not monitored for changes, and the actual values used for calculations are only set based on these parameters at certain points, notably when changed Preferences are saved or applied.

On the other hand, changing `assertRoughly` to evaluate an epsilon based on these two settings might produce false positive tests if there is actually an error but it is masked by the current Preferences. On the other (other) hand, doing this would allow tests to be run under various Preferences, avoiding the failure to ever detect an error that only occurs with non-standard Preferences.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39292 (Git)
Build type: Release
Branch: Issue16849
Hash: 15cfe9d2df30d96d6773c53a1b80ea038d11f3a0
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Classic/
```

### Subproject(s) affected?

None

### Anything else?

This was spun off #18315 since it was not relevant to the Draft module assigned to that Issue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kpemartin, 2024-12-10:
`PathTestUtils.PathTestBase` could be given a `setUp` method, called from the derived-class `setUp` methods, which evaluates a default error for `assertRoughly` based on the Preferences. It might also complain (warn) if the allowed error becomes absurdly large because of strange Preferences, though the definition of "absurdly large" is somewhat arbitrary...

For this to work, all the derived test's `setUp` methods would have to call this, but they should probably really be doing this anyway for form's sake even though the base-class method `unittest.TestCase.setUp` does nothing.

kpemartin, 2025-01-22:
Why should `PathTestBase.assertRoughly` (compare two numeric value for near-equality) and `PathTestBase.assertCoincide` (compare two points for near-equality) use different allowed error values? The former error defaults to 0.00001, the latter to 0.0001 (ten times more).

kpemartin, 2025-01-22:
Maybe trying to dynamically generate an error epsilon is not the right approach:
1. A different epsilon is used to compare the coordinates of compared (x,y,z) points than for comparing scalar values
2. This epsilon is also used, for instance, for comparing values that are not in measurement units (for instance the rotation axis for a Placement, which is a normalized unit-less direction vector)
3. The default compare epsilon of 0.00001 (mm) does not match the default Preferences settings of 0.01 mm (that is, metric measurement system with 2 precision digits). If the test specified more non-zero decimal digits on some of the numbers it uses to set up the test, it would fail even with default Preferences.

kpemartin, 2025-01-22:
I'm starting to wonder if the flaw lies entirely elsewhere. The test case is expecting, for instance, to extract a set of attributes from a `PathToolController` using `templateAttrs()`, then make a new identical `PathToolController` from these attributes using `PathToolController.FromTemplate()`.

The returned templateAttrs value is a dictionary keyed by attribute name string, with values also being strings which for distances and coordinates consist of a number and a unit. The precision in those numbers appears to be governed by the decimal digits in Preferences. The actual code uses the Python formatting operation, for instance `"%s" % (obj.VertFeed)` which ultimately uses the decimal digits Preference (as well as the units system).

I would argue that either the test should not be doing this because it is an unreliable operation at the mercy of Preferences, or `templateAttrs` (and similar methods on other objects) should be formatting the numeric values with precision suitable for round-trip conversion.

This can happen anywhere a number is converted to text and later back to a number expecting to get the original value.

For instance this means that the user can unwittingly change the value of an object's property if the precision in Preferences causes it to be rounded, the user clicks `f(x)` to bring up the Expression editor form, and quits from the form using `Ok` instead of dismissing the form with the Close "X" in the title bar or the Escape key. This will change the original value to the rounded value shown in the Expression editor form.

luzpaz, 2026-10-26:
@kpemartin any chance you can test if #30593 does in fact fix this issue ?

kpemartin, 2026-10-28:
@luzpaz working on it now. Looking at the code I expect it will solve the issue of the test failing for a problem unrelated to the feature it is trying to test.

kpemartin, 2026-10-28:
I was trying to test #30593 but was unable to build it, apparently due to the change in how the coin package is referenced in the project.

However, I can still build the head version and there I was unable to reproduce the problem. It appears that in head, the `ToolBit` objects now remember if they were created using Metric or Imperial units and so the result of `Path.Tool.Controller.templateAttrs()` in the test now returns the tool diameter in its metric form "1.750 mm" even when the app preferences are set to English units. The conversion to text is done (in this case) by `to_json(diameterValue, units="Metric")` so I don't know how it decides on three decimal places. Text conversions for path properties (as opposed to path tool properties) such as vfeed are done using `"%s".format(value)` which seems to use metric units despite Preferences, and as many decimal places as are necessary to represent the value.

So it appears that someone has fixed the underlying problem of doing the round-trip being preferences-dependent. If that is the case, the test does not have to be changed, unless one wants it to pass in another branch of the code that does not include the ToolBit changes.