FreeCAD/FreeCAD #18419: When using nested assemblies, "unconnected to a grounded joint", but it actually is via a datum

noracarmig, 2024-12-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hey ! Congrats for the 1.0.0 release.

I've noticed an issue with the Assembly workbench lately when using nested flexible assemblies.

Steps to reproduce: 

- Create file 1 with an Assembly that contains one grounded part
- Create file 2 with an Assembly that contains:
    - one grounded Part with a Local Coordinate System datum that is physically outside the part
    - one Part that has a Fixed joint to that LCS.
- Import the assembly from file 2 in the assembly from file 1
- Turn the nested assembly flexible
- Force solve file 1
-> get a warning: joint in unconnected to a grounded part so it is ignored.

This warning does not happen when the parts in file 2 are physically jointed together by a face instead of an LCS. Perhaps the round that checks the total assembly tree ignores datums ?

This issue seems pretty close to  #16436 , but It seems to have been completed before the 1.0.0 release that I'm working with so I guess my issue is a new one. Apologies if it's a duplicate.

Here is the whole tree to reproduce the issue:
[image]

Thank you

### Full version info

```shell
OS: KDE Flatpak runtime (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Installed mods: 
  * btl 0.9.9
  * kicadStepUpMod 11.4.4
  * FreeCAD-themes 2024.7.24
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-10:
can you recheck with the dev builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
@PaddleStroke FYI

maxwxyz, 2024-12-10:
Can you add an example file as .zip?

PaddleStroke, 2024-12-11:
Let us wait for https://github.com/FreeCAD/FreeCAD/pull/18332 to merge and check if this bug is still here. Because this PR is changing a lot the interaction with LCS.

noracarmig, 2024-12-11:
Here is the zip for anyone wanting to replicate on build 1.1.0dev.39560 
[AssemblyBugSituation.zip](https://github.com/user-attachments/files/18092792/AssemblyBugSituation.zip)

noracarmig, 2024-12-11:
> Let us wait for #18332 to merge and check if this bug is still here. Because this PR is changing a lot the interaction with LCS.

Noted, I'll let you know then. Thank you so much for your contribution.

noracarmig, 2024-12-14:
I just tested with the last build that includes the merged [fa8b273](https://github.com/FreeCAD/FreeCAD/pull/18332/commits/fa8b273f7d0bc0514a4fc6a97e95b73657190a67) into main. (is that the right way to refer to it ? I'm unfamiliar with git, I'm just a user).

Anyways, this fixes it. I got the migration popup, which kinda broke my model, but remaking the joints fixed it. I don't get the "unconnected to a grounded part" warning anymore when jointing a part to an LCS.

Thank you @PaddleStroke

closed 2024-12-14 by maxwxyz