FreeCAD/FreeCAD #18422: Macro Editor feature request: send selection to console

FlachyJoe, 2024-12-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I debug a macro I frequently copy-paste lines to console in order to try them. A "Send to console" command in the context menu could save time.

### Full version info

```shell
any
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-11:
cc @galou just a heads up since this macro related (hopefully not too much noise for you :sweat_smile: )

mwganson, 2024-12-25:
Related PR: https://github.com/FreeCAD/FreeCAD/pull/18716

closed 2025-01-19 by chennes

tarman3, 2025-01-31:
This is very useful features, but real behavior is little bit different
In the console will send not what is selected, but the whole line, in which the cursor is located

[image]

mwganson, 2025-01-31:
Yes, it takes the full line if any part of that line is in the selection.  We need to know the leading spaces for the first line so it can be trimmed to prevent the console giving an unexpected indentation error.  If the first line has 8 leading spaces, for example, then those 8 leadings are removed from it and also from the remaining lines in the selection.  By having all of the entire lines added to the selection we can ensure this works as expected.  It also makes it a bit easier to select since you don't have to select all the leading white space on the first line.

But it does make it more difficult for selections as made in your screenshot.  For those you can copy and paste into the console.