FreeCAD/FreeCAD #18425: Pad: up to shape crashes.

Maykeye, 2024-12-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce

0) uninstall flatpak version, delete `~/.var/app/org.freecad.FreeCAD/`, reinstall flatpak version, create  new parametric part
1) Create a new rectangle on XY plane
2) Choose pad operation
3) Select "up to shape" type

Get a crash

```console
/usr/include/c++/13.2.0/bits/stl_vector.h:1208: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::front() [with _Tp = Part::cutTopoShapeFaces; _Alloc = std::allocator<Part::cutTopoShapeFaces>; reference = Part::cutTopoShapeFaces&]: Assertion '!this->empty()' failed.
```

### Full version info

```shell
OS: KDE Flatpak runtime (fellariver/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-11:
Can not reproduce.

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.0; Silk 0.1.5; Curves 0.6.51
```
</details>

Please try with AppImage

Maykeye, 2024-12-11:
Works in appimage, doesn't work in flat or from arch arch repo

```console
 ╰ ./FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage 
fusermount3 version: 3.16.2
FreeCAD 1.0.0, Libs: 1.0.0R39109 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
Cannot access /home/fella/.local/share/FreeCAD/Mod/Freecad-Built-in-themes-beta/Classic light

~/Downloads via  v23 via 🐫 via  v3.12.7 took 22s
 ╰ org.freecad.FreeCAD
FreeCAD 1.0.0, Libs: 1.0.0R39109 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

/usr/include/c++/13.2.0/bits/stl_vector.h:1208: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::front() [with _Tp = Part::cutTopoShapeFaces; _Alloc = std::allocator<Part::cutTopoShapeFaces>; reference = Part::cutTopoShapeFaces&]: Assertion '!this->empty()' failed.

… v23 via 🐫 via  v3.12.7 took 22s
[⚡] × freecad
FreeCAD 1.0.0, Libs: 1.0.0R39109 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Cannot access /home/fella/.local/share/FreeCAD/Mod/Freecad-Built-in-themes-beta/Classic light
Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
/usr/include/c++/14.2.1/bits/stl_vector.h:1213: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::front() [with _Tp = Part::cutTopoShapeFaces; _Alloc = std::allocator<Part::cutTopoShapeFaces>; reference = Part::cutTopoShapeFaces&]: Assertion '!this->empty()' failed.
fish: Job 1, 'freecad' terminated by signal SIGABRT (Abort)
```

tarman3, 2024-12-11:
I using Archlinux.
I don't have this problem in FreeCAD from AppImage and Arch repository.

xtemp09, 2024-12-11:
Release version must be rid of asserts. Where does this flatpak come from?

----

> Looks like some problems in your config.

Sadly, config or safe mode do not influence failing asserts. ='( Something is built as debug version on his system.

Maykeye, 2024-12-11:
> Release version must be rid of asserts. Where does this flatpak come from?

Flathub, where it published by verified `flathub.org`

```console
$ flatpak list --columns=application,origin | grep -i freecad
org.freecad.FreeCAD     flathub
$ flatpak remotes -d
Name    Title   URL                          Collection ID Subset Filter Priority Options … … Homepage             Icon
flathub Flathub https://dl.flathub.org/repo/ -             -      -      1        system  … … https://flathub.org/ https://dl.flathub.org/repo/logo.svg
```

> Sadly, config or safe mode do not influence failing asserts. 

Also step 0 was the point to get rid of config entirely.  

> Something is built as debug version on his system.

Looks like it: AppImage shares config with repo, and works fine without safe mode.  Repo, flatpak versions crash in safe mode.

luzpaz, 2024-12-11:
@adrianinsaval could you transfer to freecad flatpak repo ?

Maykeye, 2024-12-11:
Also even if it is asssert, does it  needs to be there at all if getting rid of it in release  mode leaves freecad in reliable state? 

(For reference, I've stumbled on this situation, because I've misclicked and instead of "up to face"  landed on "up to shape").

xtemp09, 2024-12-11:
> Also even if it is asssert, does it needs to be there at all if getting rid of it in release mode leaves freecad in reliable state?

A program compiled in debug and in release mode are different. Optimization can remove some errors and memory leaks. The assert can come from libraries, the source code of _FreeCAD_ may be fine. Can you share a stacktrace for this crash? I've never used flatpack, so I don't know if it gives any output at all.

Maykeye, 2024-12-11:
Not sure how to get stacktrack from flatpak, but got at least something  from /usr/bin version 
```
[#0] 0x7ffff26a53f4 → mov ebx, eax
[#1] 0x7ffff264c120 → raise()
[#2] 0x7ffff26334c3 → abort()
[#3] 0x7ffff28d3af0 → std::__glibcxx_assert_fail(file=<optimized out>, line=<optimized out>, function=<optimized out>, condition=<optimized out>)
[#4] 0x7fff791101f5 → PartDesign::FeatureExtrude::makeShellFromUpToShape(Part::TopoShape, Part::TopoShape, gp_Dir)()
[#5] 0x7fff79111f30 → PartDesign::FeatureExtrude::buildExtrusion(Base::Flags<PartDesign::FeatureExtrude::ExtrudeOption>)()
[#6] 0x7ffff52ffa82 → App::DocumentObject::recompute()()
[#7] 0x7fff85afaea1 → Part::Feature::recompute()()
[#8] 0x7ffff527ec94 → App::Document::_recomputeFeature(App::DocumentObject*)()
[#9] 0x7ffff5284f4a → App::Document::recomputeFeature(App::DocumentObject*, bool)()
```

The biggest sus is at
https://github.com/FreeCAD/FreeCAD/blob/7f456d854f4e468d3893a5c3239fe257301f3c60/src/Mod/PartDesign/App/FeatureExtrude.cpp#L146

It seems if `cfaces` is still empty, release version will return `shape` as is (loop is NOP and line 156 is is essentially `if false`).
However since calling `front()` on vector is [UB](https://stackoverflow.com/questions/681725/what-do-i-get-from-front-of-empty-std-container), so it works during phase of the moon when no checks are done, but UB is bad. Probably better to check it explicitly with second `if (cfaces.empty()) return shape;`. 

Or maybe something else inlined.  Can't test as build instructions are too pyside2 dependent.

FlachyJoe, 2024-12-11:
Feel free to ping me if it's really related to the UpToShape code. :eyes:

luzpaz, 2024-12-11:
Great sleuthing @Maykeye! Looks like wwmayer submitted a fix for review in #18435 based on your investigations :pray:

closed 2024-12-13 by chennes

adrianinsaval, 2024-12-15:
I have applied the fix to the flatpak release, it should be available in a few hours https://github.com/flathub/org.freecad.FreeCAD/pull/22