FreeCAD/FreeCAD #18426: techdraw

Akimen64, 2024-12-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi, In version 1.0.0, no matter how I try to generate a view from a 3d object on a 2d drawing( isometric, from, left...), its always generating the same view on my page.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/Canada (fr_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus
  * ExplodedAssembly
  * fasteners
  * sheetmetal
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-11:
Please write steps to reproduce

https://github.com/user-attachments/assets/ed6f6ae4-8a69-4fdb-b6db-45c48e7e2b78

luzpaz, 2024-12-11:
```
  * A2plus
  * ExplodedAssembly
  * fasteners
  * sheetmetal
```
Each of those addons should have a version appended to them. Something maybe wrong with your installation perhaps? (referring to your previous closed ticket https://github.com/FreeCAD/FreeCAD/issues/18427)

maxwxyz, 2024-12-11:
Please check out how to [report issues](https://github.com/FreeCAD/FreeCAD?tab=readme-ov-file#reporting-issues) which could help us diagnose and fix it.

- share an example file with the issue present. You can upload a zipped file to the original issue.
- add steps to reproduce the issue. Ideally, provide a macro of the issue: `Macro > Macro recording...` and repeat all steps until the issue occurs.
- please change the title to reflect the issue

I can reopen the issue if you add the infos

closed 2024-12-11 by maxwxyz