FreeCAD/FreeCAD #18430: Unhiding base planes of a body causes 3d view performance degradation.

oficinerobotica, 2024-12-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the attached video take a look at the bottom left frame counter. As soon as I unhide the origin of a body , the frame counter starts measuring movement even tough there is no mouse input. This also degrades quite a bit the 3d view performance.
The problem dissapears as soon as I hide the origin planes.  

https://github.com/user-attachments/assets/c02aed36-2931-4fea-a949-ace7c5667f5b

### Full version info

```shell
OS: KDE neon 6.2 (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * fasteners 0.5.33
  * PieMenu 1.10.1
  * OpenTheme 2024.9.1
  * sheetmetal.backup1729482375.2960367 0.5.1 (Disabled)
  * sheetmetal 0.5.10
  * Curves 0.6.51
  * Gridfinity 0.6.0
  * ArchTextures
  * Assembly4 0.50.15
  * CurvedShapes 1.0.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-11:
@PaddleStroke FYI regarding core datums

PaddleStroke, 2024-12-11:
@oficinerobotica do you have the previous weekly build to test if it also happens before the core datum changes?

PaddleStroke, 2024-12-11:
@oficinerobotica can you please test if you see the same degradation if you show assembly joints? 
Both the new datum objects and joints are using 'SoShapeScale' which is scaling the size of the object to be constant. I have a feeling this might be the culprit since the rest of the core datum PR was only visual changes.

So if you can make 7 joints visible (origin has 7 datums) and compare if you have the same drop in performance please?

oficinerobotica, 2024-12-12:
@PaddleStroke I still can't find a reliable way to reproduce but it keeps happening randomly in the latest weekly build.
My observations so far

1- if the frame counter starts misbehaving on hiding/unhiding of elements restarting FC solves it. Hide/unhide of the same elements in the same file will not trigger the continuous redraw on a fresh restart. I did as instructed, unhiding joints and origin planes after a clean restart of FC but the problem didn't represent. Work for a bit on the files and it restarts with the continuous redraw

2- the continuous redraw seems to start when an assembly container is active and there are other linked files opened

3- in the video below, you can see how I can trigger the behavior (observe the frame counter) and how on hiding/unhiding of the origin planes the active window tab  jumps to the assembly file. I will try to create a simple file to share where I can trigger the continuous redraw if I can find a reliable way of reproducing.

4-Notice how the active tab jumps to a file that needs recompute after a change in a linked file. This jumping between active tabs is quite a big problem, it can happen when clicking the eye icon ,using spaceBar to hide/unhide elements, entering values in the property panel etc. 
There are times where it becomes impossible to work on a file because on each interaction with it the active tab will jump to another tab where this is linked. Does this need a separate bug report?

https://github.com/user-attachments/assets/48f6f6ff-4a72-4ffc-9571-d2e708989164

oficinerobotica, 2024-12-15:
Reliable way to reproduce on my system

1-Open the attached file ContinuousRedraw.FCStd
2-Activate the assembly by double click
3-select the "Base" part then "S,G" shortcut to open the linked part file
4-in the newly opened file tab unhide the origin of the base .... the continuous redraw starts

Some additional info , the files were created by first creating a new assembly file from the start page then using the new "Insert new part" command from the assembly workbench.

`
OS: KDE neon 6.2 (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39642 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: c80bb958f377e4d43c6a4075da173a79d153db68
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * fasteners 0.5.33
  * PieMenu 1.10.1
  * OpenTheme 2024.9.1
  * sheetmetal.backup1729482375.2960367 0.5.1 (Disabled)
  * sheetmetal 0.6.1
  * Curves 0.6.51
  * Gridfinity 0.6.0
  * ArchTextures
  * Assembly4 0.50.15
  * CurvedShapes 1.0.13
`
[ContinousRedraw.zip](https://github.com/user-attachments/files/18139163/ContinousRedraw.zip)

FEA-eng, 2025-09-14:
@oficinerobotica Is it still relevant?

luzpaz, 2026-09-12:
@paddlestroke any chance you can reproduce based on the steps in https://github.com/FreeCAD/FreeCAD/issues/18430#issuecomment-2543475523 ? (I can't test latest dev because of dependency issues, apologies)