FreeCAD/FreeCAD #18434: Beginner user improvement with "What's this?" tool

SpeeQz1, 2024-12-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Something that I have noticed is that if a user wants to find the function of the tools in FreeCAD, there are several ways, usually one of them is through the `What's this?` tool which is nestled in the Help menu, an unfortunate side effect of this is that it means it's slightly hidden and users may miss it.

What I am proposing is that to the right click context menu when pressing on a button is added the "What's this?" tool so that whenever a user right clicks on a button they could instantly see that there's an option to view what it does.

(`What's This?` location in the context menu)
[image]

Including that proposition, another thing which I noticed was the lack of videos on the wiki for each tool, if one or more videos per tool (depending on its complexity) are added to the wiki that should add another path for new users to quickly discover how a tool works and what it can do.

Perhaps even instead of directly redirecting to the wiki link in a browser, it can be made where first the video would appear in a new popup window and then the user can scroll down to see additional details for that specific action.

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * fasteners 0.5.31 (Disabled)
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * parts_library
  * Alternate_OpenSCAD 1.0.0
  * CurvedShapes 1.0.13
  * FreeCAD-Ribbon 1.4.9
  * SearchBar 1.0.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

byteofsoren, 2024-12-11:
There is already an help function builtin in FreeCAD by hovering the mouse over the icon like this.
[image]

However I agree with you that those are quite useless and unhelpful.
They should perhaps look some thing like this in the future?

[image]

Where there are an animation between each figure and an more info button?

yorikvanhavre, 2024-12-12:
Having a GIF in the tooltips would be amazingly great. However that needs some thought beforehand, because they would likely need to be produced by users on the wiki, it might be unfeasible to pack them all into the FreeCAD source, so it might need to be fetched online, etc...

I would suggest doing this in two steps, first do the simpler improvement @SpeeQz1 suggests, then later on study a way to improve the tooltip and define ways to get a gif image to display.

@SpeeQz1 mind that when buttons are greyed out (inactive), their right-click menu is disabled. However, it might still be interesting to show it. That's something you might want to have a look at as well...

hyarion, 2024-12-12:
I was looking into extending tool tips like this a while back, but there's two issues with that:
* There's a lot of actions that needs to be redesigned
* Tool tips in qt doesn't really support being designed. We would have to create our own custom tool tips for that.

Neither is a show stopper, but it's not a one man job and needs to be decided before a project like that is started, otherwise it would be alot of work that might be thrown out.

yorikvanhavre, 2024-12-12:
> Tool tips in qt doesn't really support being designed. We would have to create our own custom tool tips for that.

Indeed. We need to create/extend our own tooltip class. But I think that is done already.. (Too lazy to check)... 

AFAICS it would need support for:

* Adding buttons
* Adding images
* Checking for GIF support (with animation and transparent background)
* Allowing for images to load asynchronously (do not display while they load, or display a placeholder, and update the tooltip when they are loaded)
* Cache the loaded images on disk
* Make sure it works without internet connection (no image loading but it still works)

Then I guess we could define a format for people to produce these GIFs (suggested image size, transparent background, max number of frames, speed, where to put them on the wiki, etc), and then the tooltip could be made to look for available gifs at the appropriate location on the wiki...

But yes indeed that's not small work :sweat_smile:

yorikvanhavre, 2024-12-12:
Maybe a good way to decide on this would be to try to build some simple ones... Let me give it a try

hyarion, 2024-12-12:
FIY, this tooltip feature request (except for animation) is really similar to autodesk fusion's interface. I found this video where it can be seen: https://youtu.be/KZdyFU6_GOI?si=sddOwx09A-XtMNyU&t=169

Cropped screenshot here from video for those who don't like to follow links:
[image]

FEA-eng, 2024-12-12:
@PaddleStroke had such an idea: #9253 and I think that it would be very helpful for new users.

byteofsoren, 2024-12-12:
> @PaddleStroke had such an idea: #9253 and I think that it would be very helpful for new users.

Totally agree. FreeCAD to this day is still a bit to hard for new users due to lack of information on what each button does.

Reqrefusion, 2024-12-13:
Actually, I thought about this problem and I think I found a good solution, especially for their maintenance. Simply, the relevant tooltips will be extracted from the wiki, converted and stored in a github repo as HTML. During this conversion, if there is a gif on the page, it will be taken as a gif, otherwise it will be the first png format image on the page. The Tooltip tool will show these somehow. Since this is a repo to show offline, it can be set as an addon and can be downloaded with the help of the addon manager and always kept up to date. Actually, I did this for the entire wiki in this repo, so the heavy part is already done. 

The good thing about this is that it does not require any extra maintenance. With a few lines of code(I hope), we have very acceptable tooltips, in every language. If it is thought to be missing, it can be easily changed and always remains up to date. It completely eliminates the maintenance difficulty of the project. It combines document maintenance, which is technically an existing maintenance challenge. 

What do you think @yorikvanhavre because actually I don't have much idea about things related to display. If you need something like what I described, please tag me.

byteofsoren, 2024-12-13:
> Actually, I thought about this problem and I think I found a good solution, especially for their maintenance. Simply, the relevant tooltips will be extracted from the wiki, converted and stored in a [...]

This is the way! 

Also, instead of using gif images it would be better to use svg to better support super high dpi screens in the future.
(I still have problem with FreeCAD tree view icons being so small that I can not se them on my 42" screen)

MisterMakerNL, 2024-12-14:
We already had this in Linkstage, I also made some gifs for Linkstage.
I would suggest that we develop a system where a non-coder can fill in the info. So other people can fill it.

Reqrefusion, 2024-12-15:
> We already had this in Linkstage, I also made some gifs for Linkstage. I would suggest that we develop a system where a non-coder can fill in the info. So other people can fill it.

As I mentioned above, this can also be handled via wiki. There will be 1 place to fill, not 2.

Reqrefusion, 2024-12-22:
After some effort, I came up with something that provided the look I had in mind. This macro takes the HTML contents in the relevant folder as tooltips.
```
import os
from PySide2 import QtWidgets
import FreeCADGui

# Path to the folder containing the HTML files
html_folder = r"..\FreeCAD-Documentation-html\wiki"
images_folder = os.path.join(html_folder, "File")  # Folder where images are stored

# Get the FreeCAD main window
mw = FreeCADGui.getMainWindow()

# Helper function to set tooltip from an HTML file
def set_tooltip_from_html(action):
    object_name = action.objectName()
    html_file_path = os.path.join(html_folder, f"{object_name}.html")
    
    # Check if the HTML file exists
    if os.path.exists(html_file_path):
        with open(html_file_path, "r", encoding="utf-8") as html_file:
            html_content = html_file.read()

            # Convert relative image paths (e.g., 'src="File/...') to absolute paths
            images_path = images_folder.replace("\\", "/")  
            html_content = html_content.replace(
                'src="File/',
                f'src="{images_path}/'
            )
            action.setToolTip(html_content)

# Iterate through toolbars and update tooltips
for toolbar in mw.findChildren(QtWidgets.QToolBar):
    for action in toolbar.actions():
        if action.objectName():
            set_tooltip_from_html(action)

# Iterate through menus and update tooltips
for menu in mw.findChildren(QtWidgets.QMenu):
    for action in menu.actions():
        if action.objectName():
            set_tooltip_from_html(action)
```
All that remains is to prepare the system that will create the appropriate html files from the wiki. Of course, I think qttextbrowser shows gifs but doesn't play them. I couldn't figure out how to fix this. But it works fine for normal images. It is necessary to fine-tune the sizing.

I'm using this [repo](https://github.com/Reqrefusion/FreeCAD-Documentation-html) for testing.

byteofsoren, 2025-01-28:
> After some effort, I came up with something that provided the look I had in mind. This macro takes the HTML contents in the relevant folder as tooltips.

That is amazing.