FreeCAD/FreeCAD #18436: PartDesign: Unable to pad a hollow profile to a closed cylinder internal because the prism stops at the first frontier

luzpaz, 2024-12-11:
We still can't pad a hollow profile to a closed cylinder internal because the prism stops at the first frontier but I think all the other problems was solved.
[image]

_Originally posted by @FlachyJoe in https://github.com/FreeCAD/FreeCAD/issues/11111#issuecomment-2532904266_

luzpaz, 2024-12-11:
@FlachyJoe can you add the appropriate .FCStd file and About info of latest dev you tested on ? TIA!

FEA-eng, 2025-10-29:
> [@FlachyJoe](https://github.com/FlachyJoe) can you add the appropriate .FCStd file and About info of latest dev you tested on ? TIA!

File or steps to reproduce would indeed be appreciated.

FlachyJoe, 2025-10-29:
@FEA-eng It now occurs with a filled profile too

[PadToCylindrical.FCStd.remove.zip](https://github.com/user-attachments/files/23217192/PadToCylindrical.FCStd.remove.zip)

[image]

```
OS: Debian GNU/Linux forky/sid (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.43830 (Git)
Build date: 2025/10/03 17:20:10
Build type: Unknown
Branch: main
Hash: eba045064768f334741c54bc3bb02dbf0c766a1c
Python 3.13.7, Qt 6.9.2, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 6.9.2
shiboken 6.9.2, SMESH 7.7.1.0, xerces-c 3.2.4, IfcOpenShell 0.8.3.post2, OCC 8.0.0.rc1-783c3440b2
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
Navigation Style/Orbit Style/Rotation Mode: CAD/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/86.4394/1
```