FreeCAD/FreeCAD #18440: Copying an Assembly containing Joints with Origin Plane References throws Exception

j-hap, 2024-12-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I created as assembly with at least one PartDesign body and created assembly joints with one of the body's origin plane as reference for the joint. I want to reuse the general topology of the assembly, but modify the parts within to other dimensions. I therefore copy and paste the whole assembly including all dependencies (all boxes in the Object selection dialog are checked). When I paste the copied assembly, everything gets renamed, of course, to have unique names (and labels as well, if your settings are default). But I also get these errors in the Report view:

```
07:02:47  <Exception> Origin "Unnamed#Origin005" doesn't contain feature with role "XY_Plane"
```

I'm not sure if that error has any consequence on the function of the joint though, everything seems to work fine, but they are thrown everytime I open the document.

The distance joint I created in the first assembly uses the XY_Plane of two bodies and looks like this:

[image]

The copied one looks like this:

[image]

Which does not raise any suspicion in comparison. I deleted and recreated the joint in the copied assembly and it looks the same, but the error is gone.  I inspected the Document.xml and only some object ids and the position in the xml of the new joint changed compared to the old. Is the order in the xml the issue here? Unfortunately deleting and recreating the joints referencing an origin plane in my real model every time I copy an assembly is very tedious, so I'm hoping for a fix. Please find the example files attached.

[examples.zip](https://github.com/user-attachments/files/18107828/examples.zip)

Thank you all for the efforts you put into FreeCAD!

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-15:
Could you try in the next weekly on Monday evening/Tuesday morning: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

@PaddleStroke FYI

j-hap, 2024-12-17:
I can confirm the same behavior in
```
OS: Arch Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * parts_library
  * fasteners 0.5.31
```
both opening the example file from above or recreating it from scatch throws some "doesn't contain feature with role" exceptions.

j-hap, 2024-12-22:
I did some digging and testing and I can get rid of the exception:
In the `Document.xml` I located the `<Object>` tag of the copied distance joint below the <ObjectData> parent tag. I moved it below the object tags of both referenced planes (XY_004 and XY_005 in this case). Afterwards the document opens without errors because the roles to the planes and the OriginFeatures are assigned to all referenced origin system and components before the links to the joints are created. So the old order of children of the <ObjectData> tag is

```
<ObjectData Count="58">
    ...
    <Object name="Body003" Extensions="True">
    <Object name="GroundedJoint001">
    <Object name="Joint001">
    <Object name="Joints001" Extensions="True">
    <Object name="Origin003" Extensions="True">
    <Object name="Origin004" Extensions="True">
    <Object name="Origin005" Extensions="True">
    <Object name="XY_Plane003">
    <Object name="XY_Plane004">
    <Object name="XY_Plane005">
    <Object name="XZ_Plane003">
    <Object name="XZ_Plane004">
    <Object name="XZ_Plane005">
    <Object name="X_Axis003">
    <Object name="X_Axis004">
    <Object name="X_Axis005">
    <Object name="YZ_Plane003">
    <Object name="YZ_Plane004">
    <Object name="YZ_Plane005">
    <Object name="Y_Axis003">
    <Object name="Y_Axis004">
    <Object name="Y_Axis005">
    <Object name="Z_Axis003">
    <Object name="Z_Axis004">
    <Object name="Z_Axis005">
    <Object name="Sketch" Extensions="True">
    <Object name="Sketch001" Extensions="True">
    <Object name="Sketch002" Extensions="True">
    <Object name="Sketch003" Extensions="True">
```

and the new order is

```
  <ObjectData Count="58">
    ...
    <Object name="Body003" Extensions="True">
    <Object name="GroundedJoint001">
    <Object name="Joints001" Extensions="True">
    <Object name="Origin003" Extensions="True">
    <Object name="Origin004" Extensions="True">
    <Object name="Origin005" Extensions="True">
    <Object name="XY_Plane003">
    <Object name="XY_Plane004">
    <Object name="XY_Plane005">
    <Object name="Joint001">
    <Object name="XZ_Plane003">
    <Object name="XZ_Plane004">
    <Object name="XZ_Plane005">
    <Object name="X_Axis003">
    <Object name="X_Axis004">
    <Object name="X_Axis005">
    <Object name="YZ_Plane003">
    <Object name="YZ_Plane004">
    <Object name="YZ_Plane005">
    <Object name="Y_Axis003">
    <Object name="Y_Axis004">
    <Object name="Y_Axis005">
    <Object name="Z_Axis003">
    <Object name="Z_Axis004">
    <Object name="Z_Axis005">
    <Object name="Sketch" Extensions="True">
    <Object name="Sketch001" Extensions="True">
    <Object name="Sketch002" Extensions="True">
    <Object name="Sketch003" Extensions="True">
```

notice how the `<Object name="Joint001">` was moved down 7 positions.

FEA-eng, 2025-11-24:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20251119 (Git shallow)
Build date: 2025/11/19 23:28:17
Build type: Release
Branch: main
Hash: deab0abf8e61d69f2208b794f64ac56428467fef
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

luzpaz, 2025-11-24:
> However I think there was a reason why we stored as we do. And changing that is a really big change. So just to fix this is probably not worth it. Because this bug is relatively minor :
>
> * It happens only in kind of specific situation : copy paste of an assembly which has the object themselves inside the assembly instead of links. + the reference needs to be on the origin. + need to copy paste. So it's a very narrow (and not recommanded) usecase.
> * It's just a printout, and then it works without issue

How to proceed in that case ?