FreeCAD/FreeCAD #18446: Addon manager Crash if  "Download Macro metadata" is active

Mannshoch, 2024-12-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Addon-Manager freezing if the setting "Download Macro metadata" is active.

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39583 (Git) Snap 1223
Build type: Release
Branch: main
Hash: 7f456d854f4e468d3893a5c3239fe257301f3c60
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-13:
@chennes FYI

closed 2024-12-14 by chennes

Mannshoch, 2024-12-17:
That was fast solved. Thanks a lot. Works now.

Mannshoch, 2024-12-20:
@chennes 

I did some tests.

1. `sudo snap remove freecad --purge && sudo snap install freecad --edge && freecad`
2. After freecad start chose my settings
3. Open Addon manager
4. Click on Python dependencies
5. close Addon Manager
6. Close FreeCAD

Terminal Output:
```
freecad entfernt
freecad (edge) 1.1-g0609fa23 from FreeCAD (freecad-org) installed
[error] cannot open locale definition file `en': No such file or directory
FreeCAD 1.1.0, Libs: 1.1.0devR39681 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Adding snap-specific PYTHONPATH to sys.path: /home/USER/snap/freecad/common/.local/lib/python3.10/site-packages:/snap/freecad/1230/lib/python3.10/site-packages:/snap/freecad/1230/usr/lib/python3/dist-packages
Qt: Session management error: Could not open network socket
Migrating Start Workbench to Start command... done.
connect failed: No such file or directory
propsReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.208\" (uid=1000 pid=14149 comm=\"/snap/freecad/1230/usr/bin/FreeCAD\" label=\"snap.freecad.freecad (enforce)\") interface=\"org.freedesktop.DBus.Properties\" member=\"GetAll\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=1326 comm=\"/usr/sbin/NetworkManager --no-daemon\" label=\"unconfined\")"
nmReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.208\" (uid=1000 pid=14149 comm=\"/snap/freecad/1230/usr/bin/FreeCAD\" label=\"snap.freecad.freecad (enforce)\") interface=\"org.freedesktop.NetworkManager\" member=\"GetDevices\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=1326 comm=\"/usr/sbin/NetworkManager --no-daemon\" label=\"unconfined\")"
"Object path cannot be empty"
Invalid version specified for tag version in Addon Trails: 2022-01
pip timed out
```

Current FC Info:
```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39681 (Git) Snap 1230
Build type: Release
Branch: main
Hash: 0609fa2335b0a4a3277875a986618c226a3edbe1
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

```

PR #18472 (2024-12-14): Addon Manager: Support case of repo=None in worker
Fix #18470.
Fix #18446.