FreeCAD/FreeCAD #18454: problems importing 3D DXF

juanpc2018, 2024-12-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

most 2D import Ok.

example test files:
https://jblpro.com/en/support_downloads/download_types/caddrawing

### Full version info

```shell
...
```

### Subproject(s) affected?

File formats

### Anything else?

...

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-12-13:
The files there that are composed of explicit entities (lines, arcs,...) open fine:
[image]

Those that don't import are using ACIS solids, which is a "blob" in a proprietary format inside the otherwise open DXF file. FreeCAD cannot read ACIS blobs, as stated in the documentation.

Autodesk ACIS format is a proprietary, undocumented format, it's different from the original ACIS format (which is somewhat documented), and we have no way to support it with the means of a FOSS project like FreeCAD (it would swallow all our resources to do heavy and imperfect reverse-engineering). Unfortunately, working with proprietary formats is always a PITA...

So at the moment we have no way to solve this, your best path would be to use a proprietary app able to open those files and convert them to an open format like STEP.

I'll close this as there is no action for us to undertake, but please reopen if I misunderstood anything

closed 2024-12-13 by yorikvanhavre

juanpc2018, 2024-12-13:
IF DXF does Not work,
what works?

converting DXF to STL ?
importing STL ?