FreeCAD/FreeCAD #18456: Crash in Cross-Section

greggjaskiewicz, 2024-12-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Cross Section in Part WB crashes

### Full version info

```shell
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.39560 (Git) Conda
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * AnimationFreeCAD 1.0.0
  * A2plus 0.4.62
  * sheetmetal 0.4.0
  * freecad_metal_workbench 0.0.1
  * Curves 0.6.23
```

### Subproject(s) affected?

Part

### Anything else?

https://github.com/user-attachments/assets/0d501484-e02d-4935-ba4f-36da8398fd46
[freecad_crash.txt](https://github.com/user-attachments/files/18118307/freecad_crash.txt)

https://github.com/user-attachments/assets/aaa43e26-06ef-4955-8403-5bdb4c184c11

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-13:
please attach FCStd file

greggjaskiewicz, 2024-12-13:
Attached, sorry, I assumed it was attached - but GitHub rejects FCStd file extension, even tho it's a zip. 
[voron_2.4_x_carriage.FCStd.zip](https://github.com/user-attachments/files/18132009/voron_2.4_x_carriage.FCStd.zip)

tarman3, 2024-12-13:
Also con not create cross section, but without crash.
Error message:
`Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Base shape is not valid for boolean operation`

<details>
  <summary>FreeCAD 1.1.0dev.39560 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.1; Silk 0.1.5; Curves 0.6.51

```
</details>

luzpaz, 2024-12-23:
No crash but get the same error as @tarman3

```
pyException: Traceback (most recent call last):
File "<string>", line 1, in <module>
<class 'Part.OCCError'>: Base shape is not valid for boolean operation
20:24:56  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Base shape is not valid for boolean operation
```

```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

luzpaz, 2024-12-23:
@johnocfii any chance you can test this on macOS ?

JohnOCFII, 2024-12-23:
> @JohnOCFII any chance you can test this on macOS ?

Crashes for me as well on macOS:

```
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.6.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.33
  * Curves 0.6.52
```

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [15179]
Path:                  /Applications/Other Apps/FreeCAD-39708.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-12-22 21:12:06.4456 -0600
OS Version:            macOS 15.2 (24C101)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       7C8804E1-ED9C-470E-ADF7-C2AB7A88071D

Time Awake Since Boot: 130000 seconds
Time Since Wake:       20445 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   freecad [15179]

Application Specific Information:
abort() called

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x19b92f720 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x19b967f70 pthread_kill + 288
2   libsystem_c.dylib             	       0x19b874908 abort + 128
3   libc++abi.dylib               	       0x19b91e44c abort_message + 132
4   libc++abi.dylib               	       0x19b90ca40 demangling_terminate_handler() + 348
5   libobjc.A.dylib               	       0x19b5b53f4 _objc_terminate() + 172
6   libc++abi.dylib               	       0x19b91d710 std::__terminate(void (*)()) + 16
7   libc++abi.dylib               	       0x19b920cdc __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 88
8   libc++abi.dylib               	       0x19b920c84 __cxa_throw + 92
9   libFreeCADBase.dylib          	       0x1045e412c Base::PyException::raiseException() + 496
10  libFreeCADBase.dylib          	       0x1045e4a58 Base::InterpreterSingleton::runString(char const*) + 508
11  libFreeCADGui.dylib           	       0x106ae0244 Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) + 168
12  PartGui.so                    	       0x14f042070 PartGui::CrossSections::apply() + 1612
13  libQt5Core.5.15.15.dylib      	       0x10658ffec 0x106448000 + 1343468
14  libQt5Core.5.15.15.dylib      	       0x10657312c QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
15  libQt5Core.5.15.15.dylib      	       0x10657312c QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
16  CoreFoundation                	       0x19ba50828 __CFRunLoopDoSource0 + 176
17  CoreFoundation                	       0x19ba5058c __CFRunLoopDoSources0 + 244
18  CoreFoundation                	       0x19ba4f128 __CFRunLoopRun + 840
19  CoreFoundation                	       0x19ba4e724 CFRunLoopRunSpecific + 588
20  HIToolbox                     	       0x1a6fa6530 RunCurrentEventLoopInMode + 292
21  HIToolbox                     	       0x1a6fac17c ReceiveNextEventCommon + 216
22  HIToolbox                     	       0x1a6fac508 _BlockUntilNextEventMatchingListInModeWithFilter + 76
23  AppKit                        	       0x19f5b9034 _DPSNextEvent + 660
24  AppKit                        	       0x19ff1d2d4 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
25  AppKit                        	       0x19f5ac060 -[NSApplication run] + 480
26  libqcocoa.dylib               	       0x10a3973c0 0x10a368000 + 193472
27  freecad                       	       0x104186e6c main + 3228
28  dyld                          	       0x19b5e8274 start + 2840
```

luzpaz, 2024-12-24:
@maxwxyz we should add this to the bugfix release roadmap

greggjaskiewicz, 2024-12-24:
I do wonder if my crashes are due to some settings

kadet1090, 2025-08-13:
Warning is needed, that I'm sure. But maybe we should not block the operation that could succeed from happening.

PR #23090 (2025-08-15): Part: Prevent crash on cross-section of invalid object
This fixes crash that can happen due to uncaught exception in Cross Sections computation. It does not make the cross sections compute properly but gracefully handles that case.

## Issues
Fixes: #18456

## Before and After Images
PR #23102 (2025-08-25): Part: Relax boolean requirements
The current solution checked shapes before allowing boolean operation to happen. That meant that even small error with model in an unrelated place could prevent the boolean from being computed, even if it would compute fine and create valid shape. It seems like overkill as it can prevent users from achieving the goal. 

This commit changes that behavior so the tool at least tries to compute the results.

## Issues

Fixes: #18456 - It will actually compute fine with that patch.

## Before and After Images
closed 2025-08-15 by chennes