FreeCAD/FreeCAD #18457: FreeCad V1.0 Assembly WorkBench causing access errors

MacChalium, 2024-12-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

an imported part, a dowel pin, refuses to create a joint with a solid base that has been grounded. Trying to add a simple assembly which is supposed to rotate on this dowel pin causes the pin to move back to its insertion point in the first drawing. Backing it out of the top assembly causes error that cuases the program to be unable to save and it throws an access error fron the OSA pdf showing how the assy. is supposed to look was created by using the transform tool and is added to the . A Zip of the entire design is also provided below.

### Full version info

```shell
Full version data is copied below as the second screenshot. I copied the latest weekly release and tried to use it per the directions, but I didn't know what to do with the command prompt I got when I clicked on the FreeCad .exe file.  I am a relative newbie to FreeCad and not a Pythonista so I may not do the right things, but I have been trying to put these things together for a week now. I was trained on Solidworks 3D so I am not totally new to modeling. I am using Windows 10 on a 1-year-old Dell laptop with a ton of memory so I am not suspecting hardware errors to be causing these problems
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]
[DRILL_FIXTURE_2.zip](https://github.com/user-attachments/files/18118498/DRILL_FIXTURE_2.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-13:
Please add full about info to the original issue

MacChalium, 2024-12-13:
I noticed that the line labeled  Hash does not show the entire line that I thought I had copied here from Note Pad++. It is also colored Blue which makes me think it isn't shown complete for some other reason. I can send the full hash as I saved the clipboard to my hard drive.

luzpaz, 2024-12-18:
@PaddleStroke any chance you can reproduce ?

PaddleStroke, 2024-12-18:
Could you please try to simplify the file to the minimum and give precise reproduction steps?
Because as it is there are too many files, I don't know what file to open and what steps to do to reproduce.

PaddleStroke, 2025-07-23:
bump @MacChalium . This needs minimal example and clear reproduction steps or I can't do anything about it.
Else this needs to be closed

closed 2025-09-27 by maxwxyz