FreeCAD/FreeCAD #18458: Could not import SVG that contain multibyte sequence in it.

totake0224, 2024-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello.
FreeCad could not import SVG file that contain multibyte sequence in property named "aria-label".
it output error like this 
<class 'UnicodeDecodeError'>: 'cp932' codec can't decode byte 0x88 in position 56970: illegal multibyte sequence
(whole message is attached)
[freecad error.txt](https://github.com/user-attachments/files/18122035/freecad.error.txt)

to reproduce this. I prepare small SVG file named "multibyte_in_aria-label.svg"
[image]

and I also prepare SVG file remove multibyte in aria-label named "remove_multibyte_in_aria-label.svg"
just remove multibyte from aria-label property. and this file, freecad import finish successfully.
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Japanese/Japan (ja_JP)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

totake0224, 2024-12-13:
In addition to this.
FreeCad also output same error multibye in propety "sodipodi:docname"

maxwxyz, 2024-12-15:
How do you import it?

totake0224, 2024-12-16:
at Draft mode.
use File->import menu.
I think mode is not matter.

totake0224, 2024-12-16:
please save images at first post and check import function.
2 images look same but first one is contain multi-bye string and second doesn't.

totake0224, 2025-01-14:
I found what is the problem.
This problem may occur only Japanese Windows environment.
explanation.
InkScape output utf-8 SVG file on windows, but
Freecad use pyopen without encoding parameter in Mod/Draft/ImportSVG.py.
like
> line 1788 : parser.parse(pyopen(filename)) 

So pyopen use default encoding CP932 that is python default encoding in Japanese Windows.
then it's fail.
I fixed this by adding encoding to 'pyopen' like,
> line 1788 : parser.parse(pyopen(filename,encoding='utf-8'))
I think it's more better that detecting file charset before pyopen. but I could not find way to install python library 'chardet' which provide charset detection.

totake0224, 2025-01-15:
I think encoding='utf-8' is OK.
because I believe utf-8 is default encoding of MAC and Linux and Windows(10 and 11) GUI application.
(Somehow Japanese Windows still use CP932 on console application)
but there is another pyopen in FreeCad. So I suggest use UTF8 Mode of python . it enabled by setting environment value PYTHONUTF8=1.
I will confirm this option work or not.

I'm very happy if my work will be useful for FreeCad community.

totake0224, 2025-01-15:
PYTHONUTF8 option doesn't work.
I set PYTHONUTF8 at Windows control panel.
but I can not confirm the value on FreeCad Python console.
I'm not Python user. it difficult to me finding the problem.
so I think encoding='utf-8' is only solution so far.
Is it need posting pull request?

luzpaz, 2025-01-15:
Yes, please make a Pull Request so you can get the recognition for finding and fixing the bug :+1:

luzpaz, 2025-01-15:
@chennes should we use the same strategy for other [instances of pyopen](https://github.com/search?q=repo%3AFreeCAD%2FFreeCAD%20pyopen&type=code) (as @totake0224 has pointed out) ?

luzpaz, 2025-01-15:
Can we make a CI check or lint check for this?

@totake0224 any interest in making a 2nd PR that implements this change across the codebase?

totake0224, 2025-01-16:
I'm interested in contribute something to open-source community.
But I'm not well trained software engineer. so it will take a time to complete.
If it's okay taking a time to complete, I'd like to do it.

Frankly speaking, Yesterday, it's my first time download FreeCad source.
I now preparing build environment of FreeCad.

totake0224, 2025-01-22:
Hi,
I'm considering how to add encoding＝'utf-8' to pyopen.
and I would like to have advice from another developer.
how about this code.
original source line 64.(importSVG.py)
```
from builtins import open as pyopen
```
I changed this to
```
import builtins
#redefine pyopen as open with encoding='utf-8'
def utf8_open(file, mode='r', buffering=-1, encoding=None, errors=None, newline=None, closefd=True, opener=None):
    if encoding is None:
        encoding = 'utf-8'
    return builtins.open(file, mode, buffering, encoding, errors, newline, closefd, opener)
pyopen = utf8_open
```
I confirmed this works fine.

luzpaz, 2025-01-23:
@chennes 👆

luzpaz, 2025-01-26:
bumping again so this doesn't fall to the wayside

totake0224, 2025-01-27:
I could not get the meaning [bumping again so this doesn't fall to the wayside]
For now,　I would create pull request.

luzpaz, 2025-01-28:
@totake0224 'bump' in the context of 'bumping a ticket' is like 'poking' someone to get their attention ;)

Thanks you for the PR!

luzpaz, 2025-01-28:
@chennes should I open a separate ticket to cover the other pyopen instances per https://github.com/FreeCAD/FreeCAD/issues/18458#issuecomment-2591571659 ?

totake0224, 2025-01-29:
I will make another PR associated to this issue. after first PR would be accepted. 
Because I think it difficult to say which function will be affected by pyopen.

closed 2025-01-30 by Roy-043