FreeCAD/FreeCAD #18460: [Feauture request] Polyline tool 90 degree Arc + vertical / horizontal line -> vertical / horizontal constraint instead of 90 degree angle

M4xTheMan, 2024-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'd like to propose a feature request regarding the [Polyline tool](https://wiki.freecad.org/Sketcher_CreatePolyline). While in one of the "Arc modes", it's possible to snap the arc to increments of 45° by holding the CTRL-key. This results in a angle constraint of the arc.

When the following line is tangent to the arc and horizontal / or vertical aligned, it doesn't get such a constrain. Which makes sense, because it's alignment is already constraint by the angle constrain of the arc and the tangent constrain between arc and the line. This is how it looks at the moment:
[image]

Without using the Polyline tool, I assume that one would give the last drawn line a vertical constrain instead of constraining the arc via an angle constrain. This is what that would look like:
[image]

In such cases, I'd like the Polyline tool to change the constrains accordingly to my last screenshot.

I'm looking forward to your opinions.

Thanks!

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * LCInterlocking
  * kicadStepUpMod 11.4.1
  * Lithophane
  * lattice2 1.0.0
  * Defeaturing 1.2.2
  * Manipulator
  * fasteners 0.5.33
  * Assembly3 0.12.2
  * sheetmetal 0.5.10
  * InventorLoader 1.4.0
  * Assembly4
  * Curves 0.6.51
  * POV-Ray-Rendering
  * OpenDark 2024.3.13
  * 3DfindIT 1.2.0
  * Gridfinity 0.6.0
  * Freecad-Built-in-themes-beta 1.2.2
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-13:
The idea is to remove the 90° angle constraint when the user draws a horizontal or vertical line following the arc with an active poly line tool? I think that might only work, if the section before the arc is also either horizontal or vertical, otherwise the arc is not 90° anymore.

M4xTheMan, 2024-12-13:
Good catch! It would be needed for all angle constrains set by the Polyline tool. In general: A horizontal / vertical constrain should be used instead of an angle constrain wherever possible. Here's an example for a 45 degree angle.

[image]