FreeCAD/FreeCAD #18461: Core GUI feature request: Ctrl+Drag in the tree to copy

FlachyJoe, 2024-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A frequently found shortcut is Ctrl+Drag&Drop to copy an element, this shortcut doesn't exist yet for the object tree. Should be great to add.

### Full version info

```shell
any
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-13:
Copy only the original selection or all related dependencies?
@PaddleStroke is this feasible to integrate in the tree view dragging?

PaddleStroke, 2024-12-13:
I seem to recall that we already have this. RT introduced such behaviors. But it was kind of a mess and generating issues, especially because the special mode were happening without modifiers depending on the context. This was removed. But the special modes on modifier should still work.

I seem to recall that the modifiers where shift and alt.
However I have tested and did not find any combinaison that copy the object. I think the only copying was happening when the object could not be dragged from the parent, then it was making a copy automatically (without modifiers) which was generating issues.

Longstory short, yes it can probably be done. However because copy paste is kind of a mess I don't know how it going to be.

closed 2024-12-23 by chennes