FreeCAD/FreeCAD #18463: Core GUI feature request: scroll property item content on hover if it overflow

FlachyJoe, 2024-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When a property item overflow it currently shows 3 dots and we are unable to know the full content:
[image]

If the content scroll we will see the full content.
An alternative is to display the content in the tooltip next to the property description.

### Full version info

```shell
any
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2025-01-19:
@FlachyJoe, is there an application that utilizes the same concept? I checked _Qt Creator_, it does not have such a thing. Scrolling if it is overflowed looks really foreign.

_Qt Creator_ shows tooltips, independently if the field is overfilled or not.

FlachyJoe, 2025-01-19:
We can use the tooltip but it's already filled with the property description.