FreeCAD/FreeCAD #18464: linear pattern on wayland crashes freecad

muezabdalla, 2024-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was trying to make a linear pattern by default it choose the horizontal axis (x-axis) so I tried to change it to y-axis which make the program crash
[image]
it tries to make the pattern on the positive direction and crash before even showing the preview.
here is the output in the terminal:
`FreeCAD 1.0.0, Libs: 1.0.0R1_voidlinux
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
Transformed: Result has multiple solids. Only keeping the first.
Transformed: Result has multiple solids. Only keeping the first.
xdg_surface#56: error 1: xdg_popup has no parent
The Wayland connection experienced a fatal error: Protocol error`

**steps to reproduce the bug**
1.you need freecad version that is running on wayland natively (not xwayland)
2.extract this file below then open the file called "linear_pattern.FCStd"
3.select the last pad feature and click linear pattern
4. change the axis to y-axis

### Full version info

```shell
OS: Void Linux (labwc:wlroots/wayland)
Architecture: x86_64
Version: 1.0.0.1_voidlinux
Build type: Release
Python 3.12.7, Qt 6.7.2, Coin 4.0.0, Vtk 9.3.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * freecad.gears 1.2.0
  * CfdOF 1.25.17
```

### Subproject(s) affected?

None

### Anything else?

I tried to use the latest weekly build with the appimage format but it runs using xwayland
compiling also runs under xwayland

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-13:
Please attach file

muezabdalla, 2024-12-13:
sorry for that I forgot to add it
but it is now on the top of the issue

maxwxyz, 2024-12-13:
Cannot reproduce on:
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.3
  * CurvedShapes 1.0.13
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.1
  * Ship 2024.11.26
  * Silk 0.1.5
```

muezabdalla, 2024-12-13:
> Can not reproduce
> 
I will test it in other compositors to see if it is a problem in labwc

> Cannot reproduce on:

this bug is only on wayland you cannot reproduce it on windows or on xorg

muezabdalla, 2024-12-15:
I tried to reproduce it in wayfire and did not succeed.
also tried it the same compositor (labwc) on the same distro (void linux) but on a differante installation and also did not succeed.
so the problem is probably in my current installation.

closed 2024-12-15 by muezabdalla