FreeCAD/FreeCAD #18465: Option: Export - SVG - Translate white line color to black doesn't work when dark mode. In SVG, the lines are light.

meat-source, 2024-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi! It looks like the "Translate white line color to black" option doesn't work when I export SVG. I tried to run it in safe mode - the same thing. If you switch to a light theme, then all the lines in SVG become black, as they should be.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-13:
Please provide a demo file.
@kadet1090 the SVG imports are like draft object in that regard that they currently cannot change style automatically, right?

meat-source, 2024-12-18:
https://github.com/user-attachments/assets/a6f4ec0e-7eb2-4ddd-b30b-9a77420762c9
This is a new installation. A dark theme is selected on the welcome screen.

luzpaz, 2024-12-18:
@meat-source can you zip up the file and drag-drop it into github textbox ?

luzpaz, 2025-04-12:
@meat-source can you provide an example SVG ?

FEA-eng, 2025-09-17:
Still relevant. SVG attached below:

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

luzpaz, 2025-09-27:
Does anyone have a proposal for how to solve this issue? Where does this code even go ? Obviously it's stylesheet dependent, right ?

luzpaz, 2026-09-22:
Should be tested against https://github.com/FreeCAD/FreeCAD/pull/32827