FreeCAD/FreeCAD #18467: Core GUI feature request: multiline python panel

FlachyJoe, 2024-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The python console could greatly be expended with an editor panel in which we could type multiline code and run it on the console. The script won't be saved at run but kept in place. It will be a kind of unnamed macro.

### Full version info

```shell
any
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tholds, 2024-12-21:
The python console already supports multi-line python expressions. I just tested a for loop, a function definition and a multi-line print statement (on FreeCAD v1.0.1). It seems to be much the same as the python interpreter, but the following lines will need to be indented manually (when required).

FlachyJoe, 2024-12-26:
@tholds right but it's very inefficient in a trial-error development process.

luzpaz, 2024-12-26:
@FlachyJoe So what is the exact request here? Is there a mockup you can make to illustrate how the feature will increase benefit ?

FlachyJoe, 2024-12-26:
[image]

the **Run** button should launch the python code in the console, see #18422