FreeCAD/FreeCAD #18469: PartDesign feature request:  explode compound to bodies

FlachyJoe, 2024-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

PartDesign can create compound ie. multiple solid in one body, but doesn't offer feature to separate these solids in different bodies.

### Full version info

```shell
any
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-13:
You can use Explode compound from Part workbench

FlachyJoe, 2024-12-13:
@tarman3 yes but you next need to create bodies from the result. A PartDesign action could create the bodies in one step and with a clean tree.

tarman3, 2024-12-13:
In this case you lost history of design.
New bodies will be independent from original compounded body.

FlachyJoe, 2024-12-13:
> In this case you lost history of design.
> New bodies will be independent from original compounded body.

I don't think so

Interesting test: 

- Create a multy solid body
- Create an empty second body
- Create an empty SubShapeBinder in the second body
- Display hidden properties and add an expression on the SubShapeBinder's Shape property: `Body.Shape.Solids[1]`
- Enjoy the result

tarman3, 2024-12-13:
I can not change Shape property

https://github.com/user-attachments/assets/0ed9fca7-40c6-4e4f-af0c-4a218207bd7c

FlachyJoe, 2024-12-13:
@tarman3 RMB+Expression…

tarman3, 2024-12-13:
this is some kind of cheating ))

maxwxyz, 2024-12-13:
@kadet1090 for the Part/PartDesign merge