FreeCAD/FreeCAD #18470: Add-on Manager freezes FreeCAD: "AttributeError: 'NoneType' object has no attribute 'display_name'"

furgo16, 2024-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When refreshing the Add-on manager today, it stopped at around 40% and the whole FreeCAD UI froze. There is no way to cancel what the Add-on Manager is doing and the FreeCAD process needs to be killed to be able to use FreeCAD again

Here's the log when starting FreeCAD from the command line. There is an AppArmor error and a Python one. The Python one reports an initial error on the Trails workbench, and then the traceback starts (I'm not sure if they're related)

```python-traceback
$ freecad

FreeCAD 1.1.0, Libs: 1.1.0devR39585 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Adding snap-specific PYTHONPATH to sys.path: /home/dpm/snap/freecad/common/.local/lib/python3.10/site-packages:/snap/freecad/1224/lib/python3.10/site-packages:/snap/freecad/1224/usr/lib/python3/dist-packages
Qt: Session management error: Could not open network socket
connect failed: No such file or directory
propsReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.3639\" (uid=1000 pid=506199 comm=\"/snap/freecad/1224/usr/bin/FreeCAD \" label=\"snap.freecad.freecad (enforce)\") interface=\"org.freedesktop.DBus.Properties\" member=\"GetAll\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=2245 comm=\"/usr/sbin/NetworkManager --no-daemon \" label=\"unconfined\")"
nmReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.3639\" (uid=1000 pid=506199 comm=\"/snap/freecad/1224/usr/bin/FreeCAD \" label=\"snap.freecad.freecad (enforce)\") interface=\"org.freedesktop.NetworkManager\" member=\"GetDevices\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=2245 comm=\"/usr/sbin/NetworkManager --no-daemon \" label=\"unconfined\")"
"Object path cannot be empty"
Previous cache process was interrupted, restarting...
Invalid version specified for tag version in Addon Trails: 2022-01
Traceback (most recent call last):
  File "/snap/freecad/1224/usr/Mod/AddonManager/addonmanager_workers_startup.py", line 769, in run
    interrupted = self._process_queue(num_macros)
  File "/snap/freecad/1224/usr/Mod/AddonManager/addonmanager_workers_startup.py", line 808, in _process_queue
    self.update_and_advance(None)
  File "/snap/freecad/1224/usr/Mod/AddonManager/addonmanager_workers_startup.py", line 845, in update_and_advance
    message = translate("AddonsInstaller", "Caching {} macro").format(repo.display_name)
AttributeError: 'NoneType' object has no attribute 'display_name'
```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39585 (Git) Snap 1224
Build type: Release
Branch: main
Hash: d057fd79602acaa30a2b3e62cdac99b215cbd355
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-12-13:
I think the `Invalid version specified for tag version in Addon Trails: 2022-01` message is not related to the issue. If I hit the stop button in the Add-on Manager dialog before the progress bar reaches 40%, that message is still shown, but the UI does not freeze.

Here's the log from the command line when doing the above (stop the cache refresh before the progress reaches 40%). Notice the (unrelated?) `QObject::killTimer` warnings

<details><summary>Log</summary>

```
Invalid version specified for tag version in Addon Trails: 2022-01
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
Saved slider position at 0
```

</details>

maxwxyz, 2024-12-13:
@chennes FYI.
Cannor reproduce on
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.3
  * CurvedShapes 1.0.13
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.1
  * Ship 2024.11.26
  * Silk 0.1.5

```
Maybe related to recent issue https://github.com/FreeCAD/FreeCAD/issues/18446 ?

Neubo, 2024-12-13:
Hello,

I really don't know if my issue is related (if not, I will open another one).
With version 39560, when I open a perticular document, I get plenty of : 

`14:53:04  Access violation
14:53:06  pyException: Traceback (most recent call last):
  File "C:\_Data\Freecad\FreeCAD_Org\Mod\Draft\draftviewproviders\view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
14:53:06  pyException: Traceback (most recent call last):
  File "C:\_Data\Freecad\FreeCAD_Org\Mod\Draft\draftviewproviders\view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
14:53:06  pyException: Traceback (most recent call last):
  File "C:\_Data\Freecad\FreeCAD_Org\Mod\Draft\draftviewproviders\view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'`

that didn't happend in 39240

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * DVPScaled.brep
  * Movie 2024.2.15
  * OpenTheme 2024.9.1
  * parts_library
  * QuickMeasure 2022.10.28
  * Render.backup1733928892.9688487 2024.9.25 (Disabled)
  * woodworking 0.21.2.33771.0.23.1

PR #18472 (2024-12-14): Addon Manager: Support case of repo=None in worker
Fix #18470.
Fix #18446.
closed 2024-12-14 by chennes