FreeCAD/FreeCAD #18473: Allow 0 for patterns in transform body mode and allow booleans to ignore empty bodies

Chocolateandmilkwin, 2024-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

So for context, i am working on a "Gridfinity" style bin system for storage, and trying to make a parametric bin generator with the new variable set.
It works very well so far, but there is one problem, any bin with only size 1 on one of the axis does not work because the linear patterns i am using give errors when the variable is set to 0 (See the pictures below for context).
Futhermore i have tried making a union which includes an empty body, but that just gives the error, "Input shape is null", so event if the linear pattern would return an empty body, the union would fail.

So i propose that patterns in the "Transform body" mode, where the entire body is patterned would be allow to have 0 occurrences, which would leave the body empty.
And i propose that the boolean operations in the parts workbench would ignore empty bodies instead of returning errors.

[image]
[image]

I have also tested on 1.1.0 Dev and this is the same.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Denmark (en_DK)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Part

### Anything else?

[Bins.zip](https://github.com/user-attachments/files/18127172/Bins.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-13:
Why does you need 0 occurrences?
I set 1 and shape did not changed and without any errors.

https://github.com/user-attachments/assets/ea8532e2-ba34-461b-b1e3-33218244c147

Chocolateandmilkwin, 2024-12-13:
If you see the picture, the model is made from 9 different pieces, but in the smallest configuration only 4 of them are used.
The 5 parts that are not part of the smallest configuration are all part of linear patterns, and if the linear pattern can go to 0 and return an empty body.
You example does not cover the usecase i am describing.

tarman3, 2024-12-13:
Now clear. You want remove shape by set 0 occurrences.

tarman3, 2024-12-13:
Maybe this hack will be useful.

I added a pocket with conditional expression like `VarSet.x_amount == 0 ? 1.99 mm : 0.01 mm`
It is not working if I try to remove whole shape. I dont know why.

https://github.com/user-attachments/assets/5ceff69e-f33e-4aa8-be37-832895d9525c

Chocolateandmilkwin, 2024-12-13:
Actually i have found a different hack that is almost the same as yours tarman3.
If i make a compound (Parts workbench) of the shape that i want to disappear, and a tiny cube that i know intersect the remaining bits in the 1x1 shape, then i can use a compound filter and some math to replace the bit with the cube in the situation i don't want it.
I will post the file as an example when i have finished it.

Chocolateandmilkwin, 2024-12-13:
Ok if anyone is interested in the hack, here is the project i have implemented it in.
I still think that what i have proposed would be useful changes so the hack is not necessary.
[Parametric - Copy.zip](https://github.com/user-attachments/files/18128409/Parametric.-.Copy.zip)

maxwxyz, 2024-12-14:
@FreeCAD/cad-advisory-group what is your feedback on this issue?

pierreporte, 2024-12-14:
In Catia and Inventor, there are mechanisms to do complex things triggered by conditions. It is possible to suppress features by checking the value of a parameter. In Catia, this is called a rule. Inventor provides the option via the feature property dialog.

* http://catiadoc.free.fr/online/kwrug_C2/kwrugat1030.htm
* https://www.youtube.com/watch?v=Ie-Dg2kQGZ4

I think that providing this kind of feature to FreeCAD would be better than modifying tweaking existing features. It will be more flexible and could be part of the configuration/variant feature (for you @pieterhijma).

For the presented use case, Python could be used to generate FreeCAD files.

Chocolateandmilkwin, 2024-12-14:
[pierreporte](https://github.com/pierreporte) Thanks for the python suggestion, i have already found a workaround though.

I am unsure about you other point, Freecad already support suppression, and you can setup parameters and make rules exactly like in the youtube video you supplied.
I hooked up this test file with a similar shape and rule as the video.
[Suppress.zip](https://github.com/user-attachments/files/18135675/Suppress.zip)

I think this change make sense because it would be intuitive to new users (At least to me it makes sense that a list/pattern with the length of 0 is empty, and also it makes sense that an union of a body and an empty body simply returns the none-empty body).

I also think there would not be any conflicts with legacy documents, as right now both cases causes errors and invalid bodies, so any old documents with patterns and booleans would have had to specifically make sure to never have an occurence number of 0 and empty bodies in booleans.

pieterhijma, 2024-12-14:
Just for reference, I think I ran into this problem as well, documented [in this post](https://forum.freecad.org/viewtopic.php?p=691895#p691895), for convenience the quote:

> In Week 5 I designed a [case for batteries](https://fabacademy.org/2023/labs/waag/students/pieter-hijma/week-05/index.html#design-and-3d). I was struggling with how to approach the design properly but I think I found the best way to do it. One frustration was that I cannot make a case with just one battery because I cannot generalize a linear pattern to having only one element (if I remember correctly).

Chocolateandmilkwin, 2024-12-14:
[pieterhijma](https://github.com/pieterhijma) Yes this is exactly the same problem i ran into, thank you for a different example.

If you are insterested in a workaround that actually achieves it, i made this using the compound tool.
[Parametric - Copy.zip](https://github.com/user-attachments/files/18128409/Parametric.-.Copy.zip)

pieterhijma, 2024-12-14:
@Chocolateandmilkwin, at the time I had very little time to look into it, but I believe I looked in the code and it seemed to me that the restiction of not being able to use 0 was artificial. So, it could be that allowing 0 as a value is a simple fix.

Chocolateandmilkwin, 2024-12-14:
[pieterhijma](https://github.com/pieterhijma)
Seems like that to me too, i think the problem is that Freecad has a bunch of error messages related to empty bodies, so they might not be happy making the change if the user will see a bunch of fake error messages.
For example, the file i have uploaded works perfectly fine, but throws a bunch of error messages whenever you change the parameters.

Chocolateandmilkwin, 2024-12-14:
[FlachyJoe](https://github.com/FlachyJoe) Thank you for this suggestion, i never knew you could use expressions like that.

Though i would still hope they consider my suggestion, as i think it is more user friendly.

pierreporte, 2024-12-14:
It may be more user-friendly (though it is debatable), but what you are asking is actually to modify all features so they accept a zero dimension input. I do think that it would be easier, more powerful, and more flexible to make the suppress feature expression or equivalent mechanisms easier to find and use.

Chocolateandmilkwin, 2024-12-14:
[pierreporte](https://github.com/pierreporte)
Well i only suggest to allow the booleans tools to ignore empty/null bodies instead of throwing errors, and for the pattern tool to return an empty/null body when occurrence is set to 0 in the "Transform body" mode, (though it would make sense to me if "Transform tool shapes" mode set to 0 occurrences, would have the same effect as supressing the feature).

But i do not know the inner workings of Freecad yet, so i realize that the change could be too wide to implement.

pierreporte, 2024-12-14:
Yes but those modification would need to be consistent for all FreeCAD features. The program is already full of inconsistencies waiting to be fixed.

Chocolateandmilkwin, 2024-12-14:
Well thats fair

belinski-roman, 2024-12-22:
> [пиррепорт](https://github.com/pierreporte) Что ж, я лишь предлагаю разрешить инструментам с логическими значениями игнорировать пустые/нулевые тела вместо того, чтобы выдавать ошибки, а инструменту «Шаблон» возвращать пустое/нулевое тело, когда в режиме «Преобразовать тело» установлено значение 0 (хотя для меня было бы логично, если бы режим «Преобразовать формы инструмента» с установленным значением 0 имел тот же эффект, что и отключение функции).
> 
> Но я пока не знаю, как работает Freecad изнутри, поэтому понимаю, что изменения могут быть слишком масштабными для реализации.

In my request #18669, I made a suggestion on how best to achieve the automatic suppression of the Pattern with a zero value in the Occurrences parameter.

belinski-roman, 2024-12-22:
> [pierreporte](https://github.com/pierreporte) Well i only suggest to allow the booleans tools to ignore empty/null bodies instead of throwing errors, and for the pattern tool to return an empty/null [...]

In my request #18669, I made a suggestion on how best to achieve the automatic suppression of the Pattern with a zero value in the Occurrences parameter.