FreeCAD/FreeCAD #18475: Freecad crash on deleting all assembly joints & inserted components

graunchit, 2024-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Posted on Freecad forum  https://forum.freecad.org/viewtopic.php?t=92895

With an active assembly document open, select 'Grounded joint', shift click the last inserted component, right click and select Delete, Freecad crashes.

.

### Full version info

```shell
Tested on V1.0.0

and

OS: Ubuntu 22.04.5 LTS (XFCE/xubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Anything else?

 `

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-13:
Confirmed on the [forum thread](https://forum.freecad.org/viewtopic.php?p=797306&sid=ace1546b008bc0a396bc3565d04b7b5f#p797306) by 'theber' using:
```
OS: Arch Linux (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Installed mods: 
  * sheetmetal 0.4.26
  * BOLTSFC
  * OpenTheme 2024.9.1
  * Assembly4 0.50.15
  * Curves 0.6.48
  * Assembly3 0.12.2
  * dodo 1.0.1
  * BIM021 2021.12.0
  * fasteners 0.5.29
  * A2plus 0.4.67
```

luzpaz, 2024-12-13:
@graunchit can you zip the .FCStd file, drag-drop it into this ticket (textbox) ?

graunchit, 2024-12-13:
Files
[shelf_assembly.zip](https://github.com/user-attachments/files/18128975/shelf_assembly.zip)
[shelf_parts.zip](https://github.com/user-attachments/files/18128976/shelf_parts.zip)

luzpaz, 2024-12-23:
@PaddleStroke can you reproduce ?

closed 2025-02-07 by chennes