FreeCAD/FreeCAD #18476: object rotation not consistent to preference setting

Maecx, 2024-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When selecting "object center" under rotation mode in navigation setting, rotation on object center works only,
 when I select a point outside the object. If I select a point inside/on the object, "drag on cursor" occurs. This the case both for part and part design workbench
[image]

[image]

[image]

I expected "rotate on object center" to work, no matter the point of selection

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.60n
  * Alternate_OpenSCAD 1.0.0
  * fasteners 0.4.57
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Rexbas, 2024-12-14:
I can confirm. I am not sure if the current behavior is the intended behavior but I am considering it to be a feature. @wwmayer, if I am not mistaken, you made the first implementation of the BoundingBoxCenter mode. What do you think?

wwmayer, 2024-12-14:
I cannot remember if the change in 0eba78d4009672 was on purpose or by accident. But you can easily make this mode to always rotate around the center by replacing
```
        setRotationCenterMode(NavigationStyle::RotationCenterMode::ScenePointAtCursor |
                              NavigationStyle::RotationCenterMode::BoundingBoxCenter);

```
with
```
        setRotationCenterMode(NavigationStyle::RotationCenterMode::BoundingBoxCenter);
```

Rexbas, 2024-12-15:
Changing it to always rotate around the center makes this mode almost unusable when zoomed in to a smaller part that is further away from the object center. Especially when the object center is off screen. If a pure object center mode is really wanted by users it might be better to add it as a new mode rather than changing the existing mode.

Maecx, 2024-12-18:
Since I am not working with large parts, I have never experienced the need to rotate a part after zooming in. 

I agree, that in your scenario a rotation around the object's center would be difficult to handle.