FreeCAD/FreeCAD #18481: Blinking of the screen when hovering over a new interface element

DmitriyOsipenko, 2024-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When you hover the mouse cursor over a new interface element, whether it's a navigation cube or a workbench shortcut, the screen flashes skipping a frame. 
The problem manifests itself when switching from a discrete graphics card to an embedded one.
When switching to a discrete graphics card, everything works normally.

laptop configuration:
Windows 11 Pro 24H2 26100.2605
BIOS version - 017.010.000.031

Driver version - 23.19.21.01-240819a-407052C-AMD-Software-Adrenalin-Edition
AMD driver version for Windows is 31.0.21921.1000
Direct3D API version - 12.1
Vulkan™ API Version - 1.3.260
OpenCL™ API Version - 2.0
OpenGL® API Version - 4.6
Direct3D® Driver Version - 9.14.10.01526
Vulkan™ Driver Version - 2.0.279
OpenCL® Driver Version - 31.0.21921.1000
OpenGL® Driver Version - 24.08.230729_569461f
The 2D driver version is 8.1.1.1634
User Interface version - 2024.0819.1807.1996

NVIDIA Studio - WHQL 566.36

Processor: AMD Ryzen 5 5600H with Radeon Graphics
Integrated graphics card: AMD Radeon(TM) Graphics
Discrete graphics card: Nvidia GeForce rtx 3050 laptop GPU

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-16:
Is there a driver update available? Does it also happen with the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

privard, 2025-01-25:
I'm experiencing the same issue on Windows 11 with an Nvidia RTX 3070, latest drivers. On both the official and weekly FreeCAD builds.

FEA-eng, 2025-09-27:
@DmitriyOsipenko or @privard Can you check if it's still relevant?

luzpaz, 2025-11-24:
no response. Closing.

closed 2025-11-24 by luzpaz