FreeCAD/FreeCAD #18495: FreeCAD 1.0 installed on Windows: unable to open file with Unicode characters in Path

snooker52, 2024-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

An error accured during creation of output transcoder. Msg is:
After start FreeCAT a select Theme popup message with errors:
unable to open file 'C:/Users/tom??/AppData/Roaming/FreeCAD/SavedPreferencesPacks/user.cfg.backup'
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Czech/Czech Republic (cs_CZ)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-14:
Did you have an older version of FreeCAD installed? Can you try restarting in safe mode (Help>Restart in safe mode).

snooker52, 2024-12-14:
Older FreeCAD Version is deinstalled.

Error messages in Safe Mode:
11:28:49  C++ exception thrown (boost::filesystem::create_directories: P��stup byl odep�en [system:5]: "C:/Users/tomáš/AppData/Local/Temp/freecad-RHwZJo\UserAppData\SavedPreferencePacks\Backups", "C:/Users/tomáš/AppData/Local\Temp")
11:28:49  C++ exception thrown (boost::filesystem::create_directories: P��stup byl odep�en [system:5]: "C:/Users/tomáš/AppData/Local/Temp/freecad-RHwZJo\UserAppData\SavedPreferencePacks\Backups", "C:/Users/tomáš/AppData/Local\Temp")

maxwxyz, 2024-12-14:
Does it also happen in the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

snooker52, 2024-12-14:
No, it doesn't happen with weekly build 39560.

maxwxyz, 2024-12-14:
@hyarion any idea what PR solved this?

Syres916, 2024-12-14:
https://github.com/FreeCAD/FreeCAD/pull/18068

maxwxyz, 2024-12-14:
then I'll close it as already fixed and backported. Will be in a 1.0.1 release soon.

closed 2024-12-14 by maxwxyz