FreeCAD/FreeCAD #18497: Sketcher: Trim fails to remove edges

depthoffocus, 2024-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've looked through the issues and there are some that I feel might be related but I am not sure enough.

The trim tool in the sketcher fails to remove edges in some situations. In the attached video:

- the first scenario (two overlapping circles) trims correctly
- the second scenario (overlapping squares) will not trim at all,
- in the third, only the arc edges will trim

https://github.com/user-attachments/assets/328250ed-6626-451f-b6ad-2c17fd7d26ac

### Full version info

```shell
OS: macOS 14.7.2
Architecture: x86_64
Version: 1.1.0dev.39642 (Git) Conda
Build type: Release
Branch: main
Hash: c80bb958f377e4d43c6a4075da173a79d153db68
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * Manipulator 1.5.7
  * fasteners 0.5.33
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * Curves 0.6.51
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-14:
Confirm

<details>
  <summary>FreeCAD 1.1.0dev.39560 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.1; Silk 0.1.5; Curves 0.6.51

```
</details>

AndyS1957, 2024-12-17:
Confirm for V1.1dev 39673! In V1.0.0 (39109) it works fine.

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4

xterminos, 2024-12-18:
I also can confirm this bug for: V1.1.0dev.39673

OS: Ubuntu 24.04.1 LTS (XFCE/xubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4

This bug definitely was not there in: V1.1.0dev.39287 (appimage is still on my hdd)

OS: Ubuntu 24.04.1 LTS (XFCE/xubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39287 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b5a43eb2208b91feb76b2cc363e3956922c3ad75

So  this bug appeared somewhere between V1.1.0dev.39287 (ok) and V1.1.0dev.39673 (not ok).

Unfortunately, due to the lack of an appimage-archive, I have no idea in which version this bug showed up first.
If I should take a rough guess, I would say the bug appeared 5 up to 7 releases before V1.1.0dev.39673.

depthoffocus, 2024-12-18:
I have done some more testing (macOS-x86_64).

The bug **is** present in build **39503**.

The bug is **not** present in build 39416, which (judging by my download timestamps) I believe was the preceding one.

xterminos, 2024-12-28:
The "Sketcher-Trim-Bug" still exists in: 1.1.0dev.39747

OS: Ubuntu 24.04.1 LTS (XFCE/xubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39747 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7083362acb8529017b58ea03bf7416da0192b25b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

depthoffocus, 2025-01-07:
I have been trying to use this as a small example of how to navigate the code. I think this little problem may have crept in with @AjinkyaDahale 's changes to Sketcher in https://github.com/FreeCAD/FreeCAD/pull/15811 ?

To summarise it more concisely: I believe Sketcher is currently unable to trim straight edges in most circumstances.

depthoffocus, 2025-01-07:
The error in question generally reports something like this, but I am having trouble understanding exactly where this comes from further.

```
09:22:09  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.CADKernelError'>: Geom_TrimmedCurve::U1 == U2
09:22:09  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').trim(3,App.Vector(22.651878,23.480768,0))
09:22:09  Unnamed#Sketch: Failed to trim edge
```

AjinkyaDahale, 2025-01-07:
Thanks for pointing to this issue. I think #18665 will fix it, and if not, #18916 (though it's very much a work in progress so avoid going there yet).

hyarion, 2025-01-07:
@AjinkyaDahale if it was an easy fix, would it be possible to create a separate PR for it? (If not, I think it can wait)

AjinkyaDahale, 2025-01-07:
> @AjinkyaDahale if it was an easy fix, would it be possible to create a separate PR for it? (If not, I think it can wait)

I don't think so. However, since you've already reviewed that PR, maybe it's just better to merge it? It didn't come up in yesterday's meeting because of some more fixes elsewhere recently.

Tungee, 2025-01-15:
I have the latest 1.1 build, trim doesnt work like in 1.0 still

AjinkyaDahale, 2025-01-15:
@Tungee please wait for #18665 to be merged. Alternatively, if you're willing, I'd appreciate if you could compile it and confirm it works.

Tungee, 2025-01-16:
[AjinkyaDahale](https://github.com/AjinkyaDahale) Thank you i will wait Best Regards

depthoffocus, 2025-01-18:
The trimming starts to work in 39896 but now I will get a crash instead in certain circumstances. 

(Also I note that the cursor does not display the correct one for trimming most of the time -- only momentarily after trimming an edge, apologies if you're already aware)

https://github.com/user-attachments/assets/0a0d1e8c-9129-4649-b1ba-e1e2bd29eb14

```
OS: macOS 15.2
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.35
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * Curves 0.6.53
```

Relevant bit of a typical crash log below:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [53620]
Path:                  /Applications/FreeCAD 1.1.0-39896.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-01-18 03:29:19.2101 +0000
OS Version:            macOS 15.2 (24C101)
Report Version:        12
Bridge OS Version:     9.2 (22P2093)
Anonymous UUID:        5FF5BD02-78A8-4188-0EAB-EF410C648824

Sleep/Wake UUID:       E9E69EA6-E4DA-47CC-8E48-C8DEC91D3CFB

Time Awake Since Boot: 480000 seconds
Time Since Wake:       229675 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [53620]

VM Region Info: 0 is not in any region.  Bytes before following region: 4560850944
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10fd91000-10fdb1000    [  128K] r-x/r-x SM=COW  /Applications/FreeCAD 1.1.0-39896.app/Contents/Resources/bin/freecad

Kernel Triage:
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   SketcherGui.so                	       0x12949f9c2 SketcherGui::DrawSketchHandlerTrimming::executeCommands(Base::Vector2d) + 82
1   SketcherGui.so                	       0x12949f65d SketcherGui::DrawSketchHandlerTrimming::releaseButton(Base::Vector2d) + 13
2   SketcherGui.so                	       0x1295c0bfc SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) + 2460
3   libFreeCADGui.dylib           	       0x1126fa577 Gui::ViewProvider::eventCallback(void*, SoEventCallback*) + 503
4   libCoin.80.0.3.dylib          	       0x11549dce5 SoEventCallback::handleEvent(SoHandleEventAction*) + 197
5   libCoin.80.0.3.dylib          	       0x115289489 SoAction::traverse(SoNode*) + 105
6   libCoin.80.0.3.dylib          	       0x115454293 SoChildList::traverse(SoAction*, int, int) + 131
7   libCoin.80.0.3.dylib          	       0x1154a7b75 SoGroup::doAction(SoAction*) + 101
8   libCoin.80.0.3.dylib          	       0x1154c3658 SoSeparator::handleEvent(SoHandleEventAction*) + 40
9   libCoin.80.0.3.dylib          	       0x115289489 SoAction::traverse(SoNode*) + 105
10  libCoin.80.0.3.dylib          	       0x115454293 SoChildList::traverse(SoAction*, int, int) + 131
11  libCoin.80.0.3.dylib          	       0x1154a7b75 SoGroup::doAction(SoAction*) + 101
12  libCoin.80.0.3.dylib          	       0x1154c3658 SoSeparator::handleEvent(SoHandleEventAction*) + 40
13  libCoin.80.0.3.dylib          	       0x115289489 SoAction::traverse(SoNode*) + 105
14  libCoin.80.0.3.dylib          	       0x11529423d SoHandleEventAction::beginTraversal(SoNode*) + 157
15  libCoin.80.0.3.dylib          	       0x1152891da SoAction::apply(SoNode*) + 362
16  libCoin.80.0.3.dylib          	       0x11546f391 SoEventManager::actuallyProcessEvent(SoEvent const*) + 97
17  libFreeCADGui.dylib           	       0x11268c35c SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) + 28
18  libFreeCADGui.dylib           	       0x112693133 SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) + 131
19  libFreeCADGui.dylib           	       0x1126a0c1f Gui::NavigationStyle::processSoEvent(SoEvent const*) + 79
20  libFreeCADGui.dylib           	       0x1126a7c1a Gui::TouchpadNavigationStyle::processSoEvent(SoEvent const*) + 1418
21  libFreeCADGui.dylib           	       0x1126a0944 Gui::NavigationStyle::processEvent(SoEvent const*) + 180
22  libFreeCADGui.dylib           	       0x1126c550f Gui::View3DInventorViewer::processSoEvent(SoEvent const*) + 175
23  libFreeCADGui.dylib           	       0x112684637 SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) + 279
24  libQt5Core.5.15.15.dylib      	       0x1147bee35 QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) + 157

```

AjinkyaDahale, 2025-01-18:
> The trimming starts to work in 39896 but now I will get a crash instead in certain circumstances.
> ...

If you are able to compile, could you see if it is also present in https://github.com/FreeCAD/FreeCAD/pull/18916? I can replicate one behaviour observed in the video (multiple items being trimmed instead of just one occasionally), but not the crash. Additionally, undo gives spurious results which I need to address.
 
> (Also I note that the cursor does not display the correct one for trimming most of the time -- only momentarily after trimming an edge, apologies if you're already aware)

I cannot observe this at all. The GUI side is not touched at all in that PR, so I'm surprised this is happening.

> ```
> OS: macOS 15.2
> Architecture: x86_64
> Version: 1.1.0dev.39896 (Git) Conda
> Build type: Release
> Branch: main
> Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
> Python 3.11.11, Qt 5.15.15, Coin 4.0.3, [...]

depthoffocus, 2025-01-18:
> If you are able to compile, could you see if it is also present in [#18916](https://github.com/FreeCAD/FreeCAD/pull/18916)? I can replicate one behaviour observed in the video (multiple items being [...]

I can't compile yet (must finish a job I am on before I go down this rabbithole!) but hopefully I will be able to do this soon.

> I cannot observe this at all. The GUI side is not touched at all in that PR, so I'm surprised this is happening.

My apologies -- this turns out to be a very specific niche issue! It's caused by the macOS screen recorder that I was using while testing to try to catch a crash, so it may be out of scope for FreeCAD to fix at all. 

(I haven't seen it with other apps but I will probably test with something else written in Qt first)

Tungee, 2025-01-19:
I have no screen recorder on, but my freecad still crashes when i use trim like in your examples.

xterminos, 2025-02-11:
Wow!
I did no extensive testing, but trim seems to work again in Version: 1.1.0dev.40176 

... hmm, did throw a c++ exception at some point (without crashing) which I couldn't reproduce, but still working! :-)

Thank you!

OS: Ubuntu 24.04.1 LTS (XFCE/xubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295

hyarion, 2025-02-11:
A fix for trim was merged yesterday (https://github.com/FreeCAD/FreeCAD/pull/18916) and is now available in latest weekly (40176).

@depthoffocus can you check if this dev build works for you?

depthoffocus, 2025-02-13:
> A fix for trim was merged yesterday ([#18916](https://github.com/FreeCAD/FreeCAD/pull/18916)) and is now available in latest weekly (40176).
> 
> [@depthoffocus](https://github.com/depthoffocus) can you check if this dev build works for you?

I can still crash FreeCAD with trim.

It doesn't always happen on the first run through the test, but will usually have crashed by the third attempt. It seems more likely to fail in the case where one is using trim to delete rogue edges (where delete would do as well).

I am not sure if I have managed to get it to crash in the context of my original test -- making the overlapping squares shape. Those I can do fine. So this feels like progress.

[image]

But if you carry on and use Trim to clean up a shape entirely, as people will presumably do, you eventually get a crash.

https://github.com/user-attachments/assets/902ff519-fc0e-4674-ab23-3e38aa28b2e3

```
OS: macOS 15.3
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 72/127.75
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.5
  * Curves 0.6.55
```

Relevant bit of crash report.

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [5342]
Path:                  /Applications/FreeCAD 1.1.0-40176.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-02-13 11:38:25.2260 +0000
OS Version:            macOS 15.3 (24D60)
Report Version:        12
Bridge OS Version:     9.3 (22P3051)
Anonymous UUID:        5FF5BD02-78A8-4188-0EAB-EF410C648824

Sleep/Wake UUID:       5547CFB0-AA00-4DB2-8A1E-72747E83CF28

Time Awake Since Boot: 1000000 seconds
Time Since Wake:       252452 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
[... 18 lines cut ...]
2   SketcherGui.so                	       0x11df4c91c SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) + 2460
3   libFreeCADGui.dylib           	       0x1056e28b7 Gui::ViewProvider::eventCallback(void*, SoEventCallback*) + 503
4   libCoin.80.0.3.dylib          	       0x10848fce5 SoEventCallback::handleEvent(SoHandleEventAction*) + 197
5   libCoin.80.0.3.dylib          	       0x10827b489 SoAction::traverse(SoNode*) + 105
6   libCoin.80.0.3.dylib          	       0x108446293 SoChildList::traverse(SoAction*, int, int) + 131
7   libCoin.80.0.3.dylib          	       0x108499b75 SoGroup::doAction(SoAction*) + 101
8   libCoin.80.0.3.dylib          	       0x1084b5658 SoSeparator::handleEvent(SoHandleEventAction*) + 40
9   libCoin.80.0.3.dylib          	       0x10827b489 SoAction::traverse(SoNode*) + 105
10  libCoin.80.0.3.dylib          	       0x108446293 SoChildList::traverse(SoAction*, int, int) + 131
11  libCoin.80.0.3.dylib          	       0x108499b75 SoGroup::doAction(SoAction*) + 101
12  libCoin.80.0.3.dylib          	       0x1084b5658 SoSeparator::handleEvent(SoHandleEventAction*) + 40
13  libCoin.80.0.3.dylib          	       0x10827b489 SoAction::traverse(SoNode*) + 105
14  libCoin.80.0.3.dylib          	       0x10828623d SoHandleEventAction::beginTraversal(SoNode*) + 157
15  libCoin.80.0.3.dylib          	       0x10827b1da SoAction::apply(SoNode*) + 362
16  libCoin.80.0.3.dylib          	       0x108461391 SoEventManager::actuallyProcessEvent(SoEvent const*) + 97
17  libFreeCADGui.dylib           	       0x1056743dc SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) + 28
18  libFreeCADGui.dylib           	       0x10567b1b3 SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) + 131
19  libFreeCADGui.dylib           	       0x1056d134f Gui::NavigationStyle::processSoEvent(SoEvent const*) + 79
20  libFreeCADGui.dylib           	       0x1056d835a Gui::TouchpadNavigationStyle::processSoEvent(SoEvent const*) + 1418
21  libFreeCADGui.dylib           	       0x1056d1074 Gui::NavigationStyle::processEvent(SoEvent const*) + 180
22  libFreeCADGui.dylib           	       0x1056995ef Gui::View3DInventorViewer::processSoEvent(SoEvent const*) + 175
23  libFreeCADGui.dylib           	       0x10566c6b7 SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) + 279
24  libQt5Core.5.15.15.dylib      	       0x1077b0e35 QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) + 157

```

hyarion, 2025-02-13:
It would be great if that can be reported as a new bug.

Tungee, 2025-03-14:
Ajinkya great progress so far!

AjinkyaDahale, 2025-03-14:
Thank you for your kind words @Tungee!

depthoffocus, 2025-04-12:
> I think those should be fixed with [#20150](https://github.com/FreeCAD/FreeCAD/pull/20150), 

This has been merged, I gather. 

I've just tested and I've been unable to make the macOS x86 build 41232 crash with fairly extensive trimming (or using trim to delete).

Normally it would happen within twelve or fifteen clicks perhaps.

So from my perspective, it is both working as expected and not crashing. Thanks!

closed 2025-04-13 by hyarion