FreeCAD/FreeCAD #18498: In Part Workbench Extrude not allways works

LadyLeni, 2024-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If I want to add a triangle to a cuboid using Extrude, I can't get a solid body for the triangle. File is attached.

### Full version info

```shell
1.1.0dev
39560 (Git)
```

### Subproject(s) affected?

None

### Anything else?

[Extrude failed.zip](https://github.com/user-attachments/files/18135633/Extrude.failed.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-14:
@LadyLeni please always add the full [About](https://wiki.freecad.org/About) info. There is a time-saving reason for this. TIA

tarman3, 2024-12-14:
Wire not closed, but i don`t understand why.

[image]

<details>
  <summary>FreeCAD 1.1.0dev.39560 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.1; Silk 0.1.5; Curves 0.6.51

```
</details>

tarman3, 2024-12-14:
Workaround the error by change external geometry

[image]

tarman3, 2024-12-14:
I understand.
You created projection of external geometry but not toggle to construction geometry

https://github.com/user-attachments/assets/34fe5cc1-aa9e-4c3e-a0d8-7233cca46820

tarman3, 2024-12-14:
https://wiki.freecad.org/release_notes_1.1#Sketcher_Workbench

FEA-eng, 2025-09-27:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```