FreeCAD/FreeCAD #18500: Core GUI: can't add expression on Visibility property

FlachyJoe, 2024-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Expression can be set to the Visibility property throw python but it's not possible throw the Gui.

```
>>> App.ActiveDocument.Body.setExpression('Visibility','False')
>>> 
```

### Full version info

```shell
1.1.0dev.39604
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-12-15:
It is possible with:
`App.ActiveDocument.Body.setExpression('Visibility', '0')`

FlachyJoe, 2024-12-15:
@Roy-043 yes it is, the property is accessible throw python but there is no GUI possibility.

We can set expression to every "data" (DocumentObject) properties with RMB+Expression… but it's not possible for the "view" (ViewProvider) ones.  However Visibility is shared between DocumentObject and ViewProvider so it's usable with the ExpressionEngine.

FlachyJoe, 2024-12-16:
@Roy-043 I don't care about the View properties as they are not expression-able even throw python but Visibility is.

Roy-043, 2024-12-17:
Actually you can enter an expression in the GUI for the Visibility. But you need to use 1 and 0 instead of True and False. If you reference a boolean property instead you will see that it comes in as 1/0.

FlachyJoe, 2024-12-17:
@Roy-043 I thought the visibility was not in the Data panel, but actually it is hidden, so I can close this issue.
Thank you!

closed 2024-12-17 by FlachyJoe