FreeCAD/FreeCAD #18504: Bug: Origin Axes Placement Upgrade:  destorys file on save/reopen

Andrei-Pozolotin, 2024-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Recent change in `Origin Axes Placement Upgrade:`
[image]

seems to work ... until the file safe and reopen,

1) at which point the file produces errors on open:
```python
11:20:42  <Exception> Bad object "arkon_target#Origin010"(App::Point) linked to the Origin of "arkon_target#Body008"
```
2) and shows up as corrupt in the tree view
[image]

attached:
* original or "old format" file: `arkon-source.FCStd.zip`
* updated or save-as, "new format" file: `arkon-targetFCStd.zip`

[arkon-source.FCStd.zip](https://github.com/user-attachments/files/18136686/arkon-source.FCStd.zip)
[arkon-target.FCStd.zip](https://github.com/user-attachments/files/18136691/arkon-target.FCStd.zip)

### Full version info

```shell
OS: Arch Linux (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.1.0dev.39649 (Git)
Build type: Release
Branch: makepkg
Hash: e458526279e3cf440a8f14f2424e707983de2fc1
Python 3.12.7, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Andrei-Pozolotin, 2024-12-14:
update: revert to 'last known good' revision `39622` removes the issue:
```
Version: 1.1.0dev.39622 (Git)
Hash: f5a4ec01f8b095051355226be735104ce8eb8b44
```

tarman3, 2024-12-14:
Sorry, the my question is off topic.
How did you get the version 1.1.0dev.39649 in Archlinux?

Andrei-Pozolotin, 2024-12-14:
> How did you get the version 1.1.0dev.39649 in Archlinux?

the usual `aur` guild:

```python
import os
import pathlib

this_dir = pathlib.Path(__file__).parent.resolve()
build_dir = this_dir / "freecad-git"

if not build_dir.exists():
    os.system(f"git clone https://aur.archlinux.org/freecad-git.git {build_dir}")

os.chdir(f"{build_dir}")
os.system("echo $(pwd)")

os.system("git reset --hard")
os.system("git pull")

os.system("makepkg --clean")
os.system("makepkg --syncdeps --nocheck")
os.system("makepkg --install --noconfirm")
```

maxwxyz, 2024-12-15:
Should be fixed by https://github.com/FreeCAD/FreeCAD/pull/18515 in next weekly on Monday

Andrei-Pozolotin, 2024-12-15:
@maxwxyz 
the issue is still present as of:
```
Version: 1.1.0dev.39650 (Git)
Hash: aa228f2b329793326589b760767e48f49cb7eb1c
```
which includes https://github.com/FreeCAD/FreeCAD/pull/18515

maxwxyz, 2024-12-15:
@PaddleStroke @wwmayer FYI

Andrei-Pozolotin, 2024-12-15:
> What is module `auto_mate_1`?

some internally developed workbench

wwmayer, 2024-12-15:
The reason for the mess is caused by this PR https://github.com/FreeCAD/FreeCAD/pull/17944 because it doesn't guarantee a unique name for each object any more.

If you load arkon-source.FCStd then there are e.g. two objects with the internal name Origin010:
* the origin of the body Plan
* the origin point of the origin of the part ears

The new origin objects have an extra object (App::Point) now which will be created after having loaded a project file. That's why at this point the document looks OK. But the ambiguity occurs as soon as saving and loading the project because when reading the project then some links suddenly point to a completely different object.

The result of this is that exceptions are thrown because the wrong object is linked.

Andrei-Pozolotin, 2024-12-15:
> The reason for the mess is caused by ...

great, thank you. then the proper solution would be ? ...

PaddleStroke, 2024-12-16:
Considering the gravity of the issue (though I'm not sure if it's a corner case of if it happens with all files), the safe move would be to revert the PR asap. Unless someone find what is wrong and post a fix asap.

maxwxyz, 2024-12-16:
@hyarion can you prepare a PR for that and I'll tag it

Andrei-Pozolotin, 2024-12-16:
this crash also seems related: https://github.com/FreeCAD/FreeCAD/issues/18512

kpemartin, 2024-12-16:
PR #17944 was not intended to change when a unique name is generated for each object. It should always generate unique Names within the scope of each Document. As for Labels, it should be using the same criteria for uniqueness as the replaced code. I'll see what is happening here, I assume the new code was making a false assumption. Perhaps the "upgrading" of the document is making a new DocumentObject not in the file being loaded, and the file-load code is assuming that the Names of objects in the save file are already unique so a subsequent DocumentObject from the file has a name that clashes with the name created by the upgrade...??

closed 2024-12-16 by yorikvanhavre

wwmayer, 2024-12-16:
Here is some simplified test code to simulate loading the attached project .

The first part is restoring the origin objects from Document.xml:
```
doc = App.newDocument()
doc.addObject("Part::Feature", "Origin007")
doc.addObject("Part::Feature", "Origin")
doc.addObject("Part::Feature", "Origin008")
doc.addObject("Part::Feature", "Origin010")
doc.addObject("Part::Feature", "Origin011")
doc.addObject("Part::Feature", "Origin013")
doc.addObject("Part::Feature", "Origin016")
```
So, far everything is good. The Spans container of PiecewiseSparseIntegerSet at this point has the values (10,1), (7,2), (7,2), (7, 2), (7, 2).

For the next unique name the algorithm uses the last pair of Spans. So, the unique number in this case will be 9.

In the next step the LCS objects are finished by adding an App::Point that also uses "Origin" as base name. This is simulated with:
```
doc.addObject("Part::Feature", "Origin")
doc.ActiveObject.Name # 'Origin009' => OK
doc.addObject("Part::Feature", "Origin")
doc.ActiveObject.Name # 'Origin017' => OK
doc.addObject("Part::Feature", "Origin")
doc.ActiveObject.Name # 'Origin010' => Wrong
doc.addObject("Part::Feature", "Origin")
doc.ActiveObject.Name # 'Origin018' => OK
doc.addObject("Part::Feature", "Origin")
doc.ActiveObject.Name # 'Origin011' => Wrong
doc.addObject("Part::Feature", "Origin")
doc.ActiveObject.Name # 'Origin019' => OK
doc.addObject("Part::Feature", "Origin")
doc.ActiveObject.Name # 'Origin012' => OK
```
After adding the name Origin009 Spans has the values: (10, 1), (7, 3), (16, 1), (16, 1). (16, 1).
After adding the name Origin017 Spans has the values: (10, 1), (7,3), (7,3), (7,3), (7,3)

For the next unique element the not so unique number will be 10.

@kpemartin Can you explain the logic of the values of Spans?

Below is the code of a C++ unit test to demonstrate the failure, too:
```
TEST(BaseToolsSuite, TestUniqueName9)
{
    std::string name;
    Base::UniqueNameManager manager;
    manager.addExactName("Origin007");
    manager.addExactName("Origin");
    manager.addExactName("Origin008");
    manager.addExactName("Origin010");
    manager.addExactName("Origin011");
    manager.addExactName("Origin013");
    manager.addExactName("Origin016");

    name = manager.makeUniqueName("Origin", 3);
    manager.addExactName(name);

    name = manager.makeUniqueName("Origin", 3);
    manager.addExactName(name);

    name = manager.makeUniqueName("Origin", 3);
    EXPECT_NE(name, "Origin010");
}
```

kpemartin, 2024-12-16:
Spans is a vector indexed by the digit-count, and for each digit-count there is a set of (first, count) pairs.
Each entry in the vector represents a set of integers that have been used as numbers with as many digits as the entry's index in the vector. Each entry is a (ordered) set and there should be no overlaps between the integers represented by the various pairs (and in fact two pairs should not represent perfectly adjacent spans either).

Looking at this it would seem that I'm maintaining these conditions incorrectly. This is how Spans should update:
|Proposed name|Changed name|Spans[3]|
|---|---|---|
|Origin007||(7,1)|
|Origin||unchanged, Spans[0] changed|
|Origin008||(7,2)|
|Origin010||(7,2), (10,1)|
|Origin011||(7,2), (10,2)|
|Origin013||(7,2), (10,2), (13,1)|
|Origin016||(7,2), (10,2), (13,1), (16,1)|
|Origin|Origin009|(7,5), (13,1), (16,1)|
|Origin|Origin017|(7,5), (13,1), (16,2)|

The number picked should always be larger than any existing number with the same digit count, so Origin009 was an incorrect return, though acceptable because it is indeed unique. It should have returned Origin017.
The choice of not using the gaps (0-6, 9, 12, 14, and 15) is *only* to reproduce behaviour of the old code, as observed in the unit test.
Similarly the next calls should have returned Origin018, Origin019, and so on. But Origin017 was unique so caused no problems.
The next name returned, Origin010, was quite off-the-wall. In this case the name already existed, but addObject trusted that it would be a unique name, and inserted the DocumentObject under that name. The UniqueNameManager does not raise an error when a duplicate insertion occurs, it only fails an assertion. So at this point the Spans collection remains unchanged.

I think I have to take a closer look at how erase works on a `set<...>`. I'll run the debugger against your example to see what I got wrong.

wwmayer, 2024-12-16:
After adding the names the Spans container has the content:
Origin007: (7, 1)
Origin: (0, 1)
Origin008: (7, 2)
Origin010: (10, 1), (7, 2)
Origin011: (10, 1), (7, 2), (7, 2)
Origin013: (10, 1), (7, 2), (7, 2), (7, 2)
Origin016: (10, 1), (7, 2), (7, 2), (7, 2), (7, 2)
Origin009: (10, 1), (7, 3), (16, 1), (16, 1), (16, 1)
Origin017: (10, 1), (7,3), (7,3), (7,3), (7,3)

I wonder how it is possible that the set contains multiple identical elements. What also weird is when adding the new span (11, 1) it won't appear in the set but a duplicate of (7, 2). The same happens with (11, 1) and (16, 1) and (16, 2).

Could there something be wrong with the comparer class?

kpemartin, 2024-12-16:
On thinking about it while driving somewhere I'm wondering if I have the wrong idea of the default compare for pair<uint,uint>.
The code is expecting ordering by pair::first, but having the default compare be firstly by pair::second might explain the oddities. Can't debug beyond that in the car, no HUD showing code!
In any case there should never be any duplicates of pair::first. I don't understand how, for instance, (7,2) can be in the set multiple times as it is not a multi-set.
Or maybe I'm misunderstanding the return value from erase.

Message ID: ***@***.***>

kpemartin, 2024-12-16:
@wwmayer I just got back to my computer and tried all those addObject calls in your example (well, all but the last 4) and they all worked perfectly, giving exactly the results I expected. After Origin016, it returned Origin017, Origin018, Origin019, and Origin020. After having created Origin019, the next create was going to be Origin020, and before adding the index 20 to the Spans, the Spans were (7,2), (10,2), (13,1), (16,4) representing the numbers 7, 8, 10, 11, 13, 16, 17, 18, and 19, all as expected.

This was built using commit e698c73a, which was (in my time zone) the last commit yesterday.
Compiler is MS VS2022 17.11.2 on Windows 10, Release build of FreeCAD.

I also could not see any defect in the result of loading the attached akron-source file.
The objects named "Originxxx" (using Names, not Labels for the path in the drawing tree) are:
```
Part->Origin
Part->Origin->Origin018
Body006->Origin007
Body006->Origin007->Origin017
Part001->Origin008
Part001->Origin008->Origin019
Part001->Body008->Origin010
Part001->Body008->Origin010->Origin020
Body013->Origin016
Body013->Origin016->Origin023
Body013->Boolean->Body->Origin011
Body013->Boolean->Body->Origin011->Origin021
Body013->Boolean->Body010->Origin013
Body013->Boolean->Body010->Origin013->Origin022
```
though I did not get the document-upgrade warning, which may be from code added today, after this commit.

wwmayer, 2024-12-16:
And does this test work for you, too: `TEST(BaseToolsSuite, TestUniqueName9)`?

kpemartin, 2024-12-17:
My source doesn't have a TestUniqueName9 in it at yesterday's commit, and the main branch does not either.

My suspicion now is that my code is doing some undefined operation on/within the Spans collection and actually damaging its RB-tree, at least in certain implementations, with the Windows runtime being more robust to this. I'm reading the code again and I think I may have begin() and end() confused in spots, but it's late and I have to read it again more carefully when I'm more awake.

wwmayer, 2024-12-17:
> My source doesn't have a TestUniqueName9 in it at yesterday's commit, and the main branch does not either.

You have to copy it from my comment above.

wwmayer, 2024-12-17:
> My suspicion now is that my code is doing some undefined operation on/within the Spans collection and actually damaging its RB-tree,

To me the function object comparer looks suspicious because I don't think it correctly implements the "<" relation. With the implementation of this relation you can affect if and where an element is added.

Here, is the implementation of comparer and one thing is definitely wrong:
```
        class comparer
        {
        public:
            bool operator()(const etype& lhs, const etype& rhs) const
            {
                // The equality case here is when lhs is below and directly adjacent to rhs.
                return rhs.first - lhs.first >= lhs.second;
            }
        };
```
etype is a pair of two unsigned ints. So, in case lhs.first > rhs.first then an underflow occurs with rhs.first - lhs.first and it will be a very high number. So, the output of ">=" is the exact opposite.

You should better express the relation without using a difference but an addition:
`return lhs.second + lhs.first <= rhs.first;`

At least in my test case and the test project from above the issue is fixed. But I don't know if there are any other problems.

wwmayer, 2024-12-17:
@kpemartin When re-adding the class UniqueNameManager I suggest to do it with two PRs.
The first PR should only contain the class and the unit tests. This can be merged immediately. If the code is considered correct the second PR can be started to replace the old implementation.

Hint: Since UniqueNameManager is not a lightweight class I suggest to move it to its own source and header files, i.e. UniqueNameManager.cpp and UniqueNameManager.h.

kpemartin, 2024-12-17:
Yes, now that I'm awake I'm looking at that comparer function.
I think I originally started with ints rather than uints and the compare was correct then, and I believe that my thinking was that the addition form could overflow more easily. Of course, once changed to uints, the subtraction form was subject to overflows.
The differences between your results and mine would be due to differences in the actual compare operations used in maintaining the set contents.

kpemartin, 2024-12-17:
It will take me a bit to separate out the change into two parts. Most of it is all kind of smooshed together in one commit.
I suppose since UniqueNameManager has its own source/include files, it should also get its own file under tests as well, or can the testing remain in Tools.cpp?

wwmayer, 2024-12-17:
> I suppose since UniqueNameManager has its own source/include files, it should also get its own file under tests as well, or can the testing remain in Tools.cpp?

It's probably good to put them into a new file, too.

kpemartin, 2024-12-17:
Are the tests run by the Tests_run.exe executable also available through the -t/-r command-line options of FreeCAD.exe or FreeCADCmd.exe?

wwmayer, 2024-12-17:
No, but if you want to run all C++ unit tests you can start ctest from within your build directory.

kpemartin, 2024-12-17:
I tried ctest (and after figuring out I needed the "-C Release" option) and it failed to execute the build/tests/Release/Tests_run.exe file because dll's it needs (FreeCADApp.dll, FreeCADBase.dll, python312.dll, xerces-c_3_2.dll, plus any yet-to-be-determined transitive requirements) are not side-by-side with the .EXE nor in my search path.

In any case I have a new PR for just adding the new class and its tests.
While that's percolating, I want to proceed with the rest of the changes to actually use the new class. Should I put these on the same branch, or start a new branch based on the one for the PR, or ...? I know if I just do this on the same branch and push it, the changes will just add on the existing PR.

hyarion, 2024-12-17:
> I want to proceed with the rest of the changes to actually use the new class. Should I put these on the same branch, or start a new branch based on the one for the PR, or ...? I know if I just do this on the same branch and push it, the changes will just add on the existing PR.

I’d recommend creating one branch for the new class and its tests, and then a second branch (based on the first one) for using the new class.