FreeCAD/FreeCAD #18505: Draft/TechDraw: Linked object does not adhere to 'Line print color' set in layers manager

semhustej, 2024-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

- create rectangle
- place it in a layer, which has 'Line color' set to red and 'Line print color' set to black
- set 'Use print color' property of the layer to 'True'
- create link from the rectangle
- create TechDraw page
- select the linked rectangle and use TechDraw_DraftView to create draft view
- the rectangle on the page is red - it does not adhere to the 'Line print color'

[image]

When Link is created from a group, which has the rectangle inside of it, and TechDraw_DraftView is created from this group, the rectangle is black on the page - it takes the 'Line print color' into account. This is demonstrated in the sample file.

Sample file:
[FC-linked-draftview.FCStd.zip](https://github.com/user-attachments/files/18136986/FC-linked-draftview.FCStd.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-12-15:
In the Draft WB layers are groups, and not a property of the object. You have to put the "Rectangle - direct link" object in the layer as well.

semhustej, 2024-12-17:
> In the Draft WB layers are groups, and not a property of the object. You have to put the "Rectangle - direct link" object into the layer as well.

Thank you for explanation, that makes sense.

So when link is created from the group, that includes the rectangle, this takes over the the properties of the layer. I understand, that this is because the rectangle in the linked group is not a link, but the original rectangle object.

What still confuses me is that it is possible to hide/unhide this rectangle in the linked group independent of the base rectangle. Is this correct behavior?

https://github.com/user-attachments/assets/5423fd07-c72c-416d-abee-548ce1f00426

[test-linked-02.FCStd.zip](https://github.com/user-attachments/files/18169970/test-linked-02.FCStd.zip)

Roy-043, 2024-12-18:
> What still confuses me is that it is possible to hide/unhide this rectangle in the linked group independent of the base rectangle. Is this correct behavior?

The "Visibility" property of the object in the linked group does not change when you use the space bar, the linked group has a hidden property "Visibility List" to store this information. That explains the behavior you see, which is indeed somewhat confusing.

closed 2024-12-18 by semhustej