FreeCAD/FreeCAD #18506: Measurement: Crash when measuring Part Line for the second time

FEA-eng, 2024-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Open the file: [measure_crash.FCStd.zip](https://github.com/user-attachments/files/18137062/measure_crash.FCStd.zip)
2. Activate the Measure tool.
3. Click on the line. It's correctly measured.
4. Close the Measure tool.
5. Repeat - FreeCAD crashes.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39646 (Git) Conda
Build type: Release
Branch: main
Hash: b25b83c1a58bf0ae95cbd502d0ccae27e8127cad
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Measurement

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-14:
Can`t confirm 

https://github.com/user-attachments/assets/fd273401-4b6a-4f6a-8244-cf32d87d7568

<details>
  <summary>FreeCAD 1.1.0dev.39560 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.1; Silk 0.1.5; Curves 0.6.51

```
</details>

xtemp09, 2024-12-14:
Here is the backtrace.

<details>

<summary>Backtrace

</summary>

```ruby
Thread 1 received signal SIGSEGV, Segmentation fault.
0x00007fff79b2e750 in FreeCADApp!?unregisterLabel@Document@App@@QEAAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z () from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\FreeCADApp.dll
(gdb) bt
#0  0x00007fff79b2e750 in FreeCADApp!?unregisterLabel@Document@App@@QEAAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z () from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\FreeCADApp.dll
#1  0x00007fff79b25691 in FreeCADApp!?removeObject@Document@App@@QEAAXPEBD@Z ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\FreeCADApp.dll
#2  0x00007fff45036a3d in MeasureGui!??_FStandard_OutOfMemory@@QEAAXXZ ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\lib\MeasureGui.pyd
#3  0x00007fff450369b9 in MeasureGui!??_FStandard_OutOfMemory@@QEAAXXZ ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\lib\MeasureGui.pyd
#4  0x00007fff73e56a0c in FreeCADGui!?reject@TaskView@1Gui@@IEAAXXZ ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\FreeCADGui.dll
#5  0x00007fff7a4e597d in Qt5Core_conda!?qt_static_metacall@QObject@@CAXPEAV1@W4Call@QMetaObject@@HPEAPEAX@Z ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Core_conda.dll
#6  0x00007fff795f1d14 in Qt5Widgets_conda!?qt_static_metacall@QDialogButtonBox@@CAXPEAVQObject@@W4Call@QMetaObject@@HPEAPEAX@Z () from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Widgets_conda.dll
#7  0x00007fff7a4e59d8 in Qt5Core_conda!?qt_static_metacall@QObject@@CAXPEAV1@W4Call@QMetaObject@@HPEAPEAX@Z ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Core_conda.dll
#8  0x00007fff7955ebc7 in Qt5Widgets_conda!?clicked@QAbstractButton@@QEAAX_N@Z ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Widgets_conda.dll
#9  0x00007fff7955eae8 in Qt5Widgets_conda!?click@QAbstractButton@@QEAAXXZ ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Widgets_conda.dll
#10 0x00007fff7955f7ed in Qt5Widgets_conda!?mouseReleaseEvent@QAbstractButton@@MEAAXPEAVQMouseEvent@@@Z ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Widgets_conda.dll
#11 0x00007fff794b7952 in Qt5Widgets_conda!?event@QWidget@@MEAA_NPEAVQEvent@@@Z ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Widgets_conda.dll
#12 0x00007fff79494941 in Qt5Widgets_conda!?notify_helper@QApplicationPrivate@@QEAA_NPEAVQObject@@PEAVQEvent@@@Z ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Widgets_conda.dll
#13 0x00007fff79492870 in Qt5Widgets_conda!?notify@QApplication@@UEAA_NPEAVQObject@@PEAVQEvent@@@Z ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Widgets_conda.dll
#14 0x00007fff73a69ad2 in FreeCADGui!?textChanged2@ExpressionTextEdit@Gui@@QEAAXVQString@@H@Z ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\FreeCADGui.dll
#15 0x00007fff7a4c65fb in Qt5Core_conda!?notifyInternal2@QCoreApplication@@CA_NPEAVQObject@@PEAVQEvent@@@Z ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Core_conda.dll
#16 0x00007fff79495b45 in Qt5Widgets_conda!?sendMouseEvent@QApplicationPrivate@@SA_NPEAVQWidget@@PEAVQMouseEvent@@00PEAPEAV2@AEAV?$QPointer@VQWidget@@@@_N4@Z ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Widgets_conda.dll
#17 0x00007fff794e0347 in Qt5Widgets_conda!??0QSizePolicy@@QEAA@XZ ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Widgets_conda.dll
#18 0x00007fff794de2be in Qt5Widgets_conda!??0QSizePolicy@@QEAA@XZ ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Widgets_conda.dll
#19 0x00007fff79494941 in Qt5Widgets_conda!?notify_helper@QApplicationPrivate@@QEAA_NPEAVQObject@@PEAVQEvent@@@Z ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Widgets_conda.dll
#20 0x00007fff794939ca in Qt5Widgets_conda!?notify@QApplication@@UEAA_NPEAVQObject@@PEAVQEvent@@@Z ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Widgets_conda.dll
#21 0x00007fff73a69ad2 in FreeCADGui!?textChanged2@ExpressionTextEdit@Gui@@QEAAXVQString@@H@Z ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\FreeCADGui.dll
#22 0x00007fff7a4c65fb in Qt5Core_conda!?notifyInternal2@QCoreApplication@@CA_NPEAVQObject@@PEAVQEvent@@@Z ()
   from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Core_conda.dll
#23 0x00007fff78e84df1 in Qt5Gui_conda!?processMouseEvent@QGuiApplicationPrivate@@SAXPEAVMouseEvent@QWindowSystemInterfacePrivate@@@Z () from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Gui_conda.dll
#24 0x00007fff78e6ef30 in Qt5Gui_conda!?sendWindowSystemEvents@QWindowSystemInterface@@SA_NV?$QFlags@W4ProcessEventsFlag@QEventLoop@@@@@Z () from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Gui_conda.dll
#25 0x00007fff7a50f764 in Qt5Core_conda!?processEvents@QEventDispatcherWin32@@UEAA_NV?$QFlags@W4ProcessEventsFlag@QEventLoop@@@@@Z () from E:\temp\FreeCAD_weekly-builds-39646-conda-Windows-x86_64-py311\bin\Qt5Core_conda.dll
[... cut ...]
#35 0x00007fffad7d7374 in KERNEL32!BaseThreadInitThunk () from C:\Windows\System32\kernel32.dll
#36 0x00007fffaf7dcc91 in ntdll!RtlUserThreadStart () from C:\Windows\SYSTEM32\ntdll.dll
#37 0x0000000000000000 in ?? ()
(gdb)
```

</details>

```ruby
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39646 (Git) Conda
Build type: Release
Branch: main
Hash: b25b83c1a58bf0ae95cbd502d0ccae27e8127cad
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

luzpaz, 2024-12-14:
I can confirm but have no backtrace. The only output is:  
```
/tmp/.mount_FreeCAcEMmAO/AppRun: line 31:  6632 Segmentation fault      (core dumped) "${MAIN}" "$@"
```

OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39642 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: c80bb958f377e4d43c6a4075da173a79d153db68
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

Syres916, 2024-12-15:
If someone has already started a Git Bisect, please advise, I'll start one tomorrow morning but at present I can only narrow it down to 86 commits so it'll take most of the day.

Syres916, 2024-12-16:
The root cause was [commit](https://github.com/FreeCAD/FreeCAD/commit/83202d8ad67a719cb62e29d9dc3496a076999f1f) , so by logic the merging of the [revert](https://github.com/FreeCAD/FreeCAD/pull/18546) PR should fix this.