FreeCAD/FreeCAD #18509: Consider switching main viewport from Gouraud (per-vertex) lighting to Phong (per-pixel) lighting

AnyOldName3, 2024-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As demonstrated by the animated GIF in the Wikipedia article https://en.wikipedia.org/wiki/Gouraud_shading, per-vertex lighting looks terrible on curved shiny surfaces unless the vertex density is very high, typically higher than you'd need for the shape to look right or export for 3D printing etc..

The same geometry with per-pixel lighting (also known as Phong Shading, which is a separate thing to the Phong Reflectance Model that FreeCAD already uses) typically looks significantly better. Instead of interpolating the lit colour between vertices, the normal is interpolated instead, which works better as while a specular highlight in the middle of a face can be completely missed by interpolating colours, interpolated normals would still capture the highlight.

It looks like this has been an out-of-the-box feature of Coin3D for a long time, but I'm not familiar with how FreeCAD uses their API, so don't know if this would be a five-minute one-line fix, or whether it's blocked by a mountain of other changes that'd have to happen first. Hopefully, this terrible-looking spring is enough to inspire someone:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Anything else?

I've put *None* for subproject(s) affected as I don't know which, if any, of the items in the list are responsible.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-15:
@shaise @kadet1090 FYI, is this feasible and sensible or would performance decrease?

shaise, 2024-12-15:
I would say that in today's HW even simple graphics chips can handle phong with no problems at all. Coin 4.0.3 supports shaders and phong rendering

AnyOldName3, 2024-12-15:
I'd probably go as far as to say simple GPUs from ten or fifteen years ago could, too, but I'd give that the caveat that the GPUs in phones from that long ago would potentially struggle. I don't imagine that phones from fifteen years ago are FreeCAD's highest priority in terms of hardware support.

maxwxyz, 2024-12-16:
@hyarion also FYI considering your thread on Discord

maxwxyz, 2026-11-07:
Part of this is in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/766a70161117ad7696fc273ebc4db619970b4b3b

Enhanced rendering does the lighting per pixel for Part and Mesh faces. It is optional, and cases such as textured or emissive faces and unsupported OpenGL contexts fall back to Classic.