FreeCAD/FreeCAD #18511: Draft/TechDraw: Linked Draft_Dimension is not visible in TechDraw_DraftView

semhustej, 2024-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When link (Std_LinkMake) is created from Draft_Dimension object and this object is placed on a TechDraw page via TechDraw_DraftView command, this Draft_Dimension is not visible on the page.

[image]

I also tested Draft_Text with the same result. I did not test Draft_Label, but I would expect the same behavior.

Sample file:
[FC-linked-dimension-draftview.FCStd.zip](https://github.com/user-attachments/files/18137175/FC-linked-dimension-draftview.FCStd.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-12-14:
I can reproduce the same behavior in FreeCAD 1.0.0

Roy-043, 2025-01-28:
Duplicate of #19199.

closed 2025-01-28 by Roy-043