FreeCAD/FreeCAD #18512: Draft/TechDraw: SIGSEGV, Segmentation fault.

Andrei-Pozolotin, 2024-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Random crashes when creating `TechDraw` pages from `python` code.
The crashes are "random enough": can not isolate a simple use case so far.

Sample crash trace:
```cpp
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x799b6ba4c1d0]
#1  0x799b6bcc787e in std::_Rb_tree_decrement(std::_Rb_tree_node_base const*) from /usr/lib/libstdc++.so.6+0xe
#2  0x799b6e1e8763 in Base::UniqueNameManager::PiecewiseSparseIntegerSet::Add(unsigned int) from /usr/lib/freecad/lib/libFreeCADBase.so+0x133
#3  0x799b6e1ecc35 in Base::UniqueNameManager::addExactName(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) from /usr/lib/freecad/lib/libFreeCADBase.so+0x235
#4  0x799b6e6fee5d in App::DocumentObject::onChanged(App::Property const*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x38d
#5  0x799b6e84c825 in App::Property::hasSetValue() from /usr/lib/freecad/lib/libFreeCADApp.so+0x135
#6  0x799b6e8dd64f in App::PropertyString::setValue(char const*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x10f
#7  0x799b6e8d6793 in App::PropertyString::setPyObject(_object*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x83
#8  0x799b27569722 in TechDraw::DrawSVGTemplatePy::setCustomAttributes(char const*, _object*) from /usr/lib/freecad/lib/TechDraw.so+0x72
#9  0x799b27569aa3 in TechDraw::DrawSVGTemplatePy::_setattr(char const*, _object*) from /usr/lib/freecad/lib/TechDraw.so+0x13
#10  0x799b6e1cbecb in Base::PyObjectBase::__setattro(_object*, _object*, _object*) from /usr/lib/freecad/lib/libFreeCADBase.so+0x11b
#11  /usr/lib/libpython3.12.so.1.0(PyObject_SetAttr+0x1fc) [0x799b6d977b1c]
#12  /usr/lib/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x4dd0) [0x799b6d98f820]
#13  /usr/lib/libpyside6.cpython-312-x86_64-linux-gnu.so.6.8(+0x10e09) [0x799b6dfcde09]
#14  0x799b6dfd089c in PySide::SignalManager::callPythonMetaMethod(QList<QByteArray> const&, char const*, void**, _object*) from /usr/lib/libpyside6.cpython-312-x86_64-linux-gnu.so.6.8+0x4c
#15  /usr/lib/libpyside6.cpython-312-x86_64-linux-gnu.so.6.8(+0x13a3e) [0x799b6dfd0a3e]
#16  /usr/lib/libpyside6.cpython-312-x86_64-linux-gnu.so.6.8(+0x1a835) [0x799b6dfd7835]
#17  /usr/lib/libQt6Core.so.6(+0x1b1b29) [0x799b6c1b1b29]
#18  0x799b6c1bab35 in QTimer::timerEvent(QTimerEvent*) from /usr/lib/libQt6Core.so.6+0xa5
#19  /usr/lib/python3.12/site-packages/PySide6/QtCore.cpython-312-x86_64-linux-gnu.so(+0x303779) [0x799b50703779]
#20  0x799b6c1a2d49 in QObject::event(QEvent*) from /usr/lib/libQt6Core.so.6+0x109
#21  /usr/lib/python3.12/site-packages/PySide6/QtCore.cpython-312-x86_64-linux-gnu.so(+0x2fd63a) [0x799b506fd63a]
#22  0x799b6d0fe78a in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8a
#23  0x799b6f2a6359 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xc9
#24  0x799b6c157ac8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x1a8
#25  0x799b6c2d6b1f in QTimerInfoList::activateTimers() from /usr/lib/libQt6Core.so.6+0x5ef
#26  /usr/lib/libQt6Core.so.6(+0x3c4f09) [0x799b6c3c4f09]
#27  /usr/lib/libglib-2.0.so.0(+0x5d559) [0x799b6a90d559]
#28  /usr/lib/libglib-2.0.so.0(+0xc0257) [0x799b6a970257]
#29  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x799b6a90ca55]
#30  0x799b6c3c22fd in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x8d
#31  0x799b6c161ee6 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x226
#32  0x799b6c158f4f in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0x9f
#33  0x799b6f1bbd9c in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0xa1c
#34  FreeCAD(+0x7ab2) [0x55c76ecf2ab2]
#35  /usr/lib/libc.so.6(+0x25e08) [0x799b6ba34e08]
#36  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x799b6ba34ecc]
#37  FreeCAD(+0x7e35) [0x55c76ecf2e35]
```

### Full version info

```shell
OS: Arch Linux (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.1.0dev.39622 (Git)
Build type: Release
Branch: makepkg
Hash: f5a4ec01f8b095051355226be735104ce8eb8b44
Python 3.12.7, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Andrei-Pozolotin, 2024-12-14:
possibly relevant: https://github.com/FreeCAD/FreeCAD/pull/17944

Andrei-Pozolotin, 2024-12-15:
update: 
 
1. the original crash is still "random", but we found a way to "prevent" it 100% as shown below;
in other words, mere setting of a `Label` on the `TechDraw.DrawSVGTemplate` object results in `SIGSEGV`.

2. we are not clear why :
* `DrawSVGTemplatePy::setCustomAttributes() ` would be used for a `Label`?
* how come `Label` assignment invokes a `UniqueNameManager::addExactName()`?

```python
    @classmethod
    def make_draw_page(self,
            stitch_origin:FreeCAD.DocumentObject,
            page_label:str,
            template_type:StitchPageType
        ) -> "TechDraw.DrawPage":
        use_trigger_issue_18512 = False
        this_doc = stitch_origin.Document
        draw_template:TechDraw.DrawSVGTemplate = this_doc.addObject(# type:ignore
            type=type_consts.TechDraw.DrawSVGTemplate, name=StitchName.Stitch_Template,
        )
        draw_template.Template = str(template_type.page_path)
        if use_trigger_issue_18512:
            draw_template.Label = template_type.type_name
        draw_page:TechDraw.DrawPage = this_doc.addObject(# type:ignore
            type=type_consts.TechDraw.DrawPage, name=StitchName.Stitch_Page,
        )
        draw_page.KeepUpdated = True
        draw_page.Template = draw_template
        draw_page.Label = page_label
        return draw_page
```

maxwxyz, 2024-12-15:
@WandererFan FYI

WandererFan, 2024-12-15:
I would say there is more going on than just assigning a value to Label.  That seems to work in the console and in a simple macro:

```
>>> # obj = doc.getObject("Template")
>>> ### End command Std_SendToPythonConsole
>>> obj.Label
'Template'
>>> obj.Label = "NotTemplate"
>>> obj.Label
'NotTemplate'
>>> 
```

```
# simple tester for page/template creation & Label setting
path = os.path.dirname(os.path.abspath(__file__))
templateFileSpec = path + "/TestTemplate.svg"
doc = FreeCAD.ActiveDocument

myPage = doc.addObject("TechDraw::DrawPage", "Page")
myTemplate = doc.addObject("TechDraw::DrawSVGTemplate", "Template")
myTemplate.Template = templateFileSpec
myPage.Template = myTemplate

myTemplate.Label = "NewLabelText"
FreeCAD.ActiveDocument.recompute()
```

WandererFan, 2024-12-15:
I don't know anything about UniqueNameManager.  That is something that happens deep inside App and TD never invokes it directly.

FWIW, changing the Label from the console and changing it in the macro above both invoke setCustomAttributes.

Andrei-Pozolotin, 2024-12-15:
@WandererFan 

based on some more experiments, we can add:
* yes, single simple test does not produce a `SIGSEGV`
* it takes about 10...100  `addObject(TechDraw:...)` burst to trigger a `TechDraw` crash
* even then it is not predictable, the bigger the size of the burst the higher the crash probability
* finally,  the crash seems to need `QT` event loop running, so perhaps `UniqueNameManager` has a concurrency issue

kpemartin, 2024-12-16:
I added UniqueNameManager recently to substantially speed up the generation of Names for objects. In addition, if the Preferences select that Labels should be unique, it is also involved in Label setting. That's what it is doing there.

I will look into what the problem is.

The clue about the QT event loop seems important, and unfortunately makes it terribly hard to reproduce.

Andrei-Pozolotin, 2024-12-16:
update:

 the crash is still reproducible regardless of the `on/off` state 
of the preference: `General/Document/Allow duplicate object labels in one document`

kpemartin, 2024-12-16:
The state of the `General/Document/Allow duplicate object labels in one document` preference has no effect because the Label collection is maintained whether this is on or not. The collection is maintained because this preference can be changed at any time, and having to create the collection only when a unique label is required would defeat the point of having the collection. Note that certain labelled object can also force unique labels regardless of this preference by overriding the `allowDuplicateLabels` method.

Note that because Labels are not *always* unique the Document uses another collection along with the UniqueNameManager to track Labels despite the existence of duplicates.

Andrei-Pozolotin, 2024-12-16:
> perhaps the code is trying to translate the Label (to the user's selected language) in a callback that can happen asynchronously?

* no, nothing like that in this section of code
* and yes, `QtCore.QCoreApplication.processEvents()` invocations do happen elsewhere
* but then again, "here or there" is pure `python`, which is supposed to `single-threaded-via-gil` on a `cpp` level unless there are explicit `python` calls to external `i/o`, which again is not  the case in this particular use case
* some slides about gil: http://www.dabeaz.com/python/GIL.pdf

kpemartin, 2024-12-16:
If the python path is single-threaded that leaves me still trying to determine how the QT event loop is involved, then.
I might not need to actually reproduce the crash; I might be able to place a conditional (being in the wrong thread) breakpoint and find how the unique-name code might be re-entered.
Right now, I'm re-building, which takes the better part of an hour...

Andrei-Pozolotin, 2024-12-16:
@kpemartin :  two more vague pointers:

A )`DrawSVGTemplate.Label = ...` happens as part of bigger object creation set, 
with lots of various `.Label = ...`, but only `DrawSVGTemplate.Label = ... ` triggers a crash:

<details>
<summary>collection of snippets for `.Label = ...`</summary>

```python
# snip
            target_skin:Part.Feature = this_doc.addObject(# type:ignore
                type_consts.Part.Feature, StitchName.Stitch_Face
            )
            target_skin.Label = f"Skin@{face_name}"
# snip
            target_mesh:Mesh.Feature = this_doc.addObject(# type:ignore
                type_consts.Mesh.Feature, StitchName.Stitch_Mesh,
            )
           target_mesh.Label = f"Mesh@{face_name}"
# snip
        target_wire:Part.Feature = this_doc.addObject(# type:ignore
            type_consts.Part.Feature, StitchName.Stitch_Wire,
        )
        target_wire.Label = f"Wire@{face_name}"
# snip
        draw_template:TechDraw.DrawSVGTemplate = this_doc.addObject(# type:ignore
            type=type_consts.TechDraw.DrawSVGTemplate, name=StitchName.Stitch_Template,
        )
       draw_template.Label = template_type.type_name
# snip
        draw_page:TechDraw.DrawPage = this_doc.addObject(# type:ignore
            type=type_consts.TechDraw.DrawPage, name=StitchName.Stitch_Page,
        )
        draw_page.Label = page_label
# snip
        draw_view:TechDraw.DrawViewPart = this_doc.addObject(# type:ignore
            type_consts.TechDraw.DrawViewPart, StitchName.Stitch_View,
        )
        draw_view.Label = f"View@{view_label}"
# snip
```
</details>

B ) there is some concurrency happening in some of the `TechDraw` objects:
https://forum.freecad.org/viewtopic.php?t=86850
@WandererFan : `... threads started as a result of doc.recompute() are not finished before ...`

kpemartin, 2024-12-16:
@Andrei-Pozolotin  While waiting for my build, I'm just trying to find where your code snippets are from. A text search of my entire Git tree yields nothing for `make_draw_page` nor for `target_wire`

Andrei-Pozolotin, 2024-12-16:
> trying to find where your code snippets are from

this is from internally developed workbench which is still private

closed 2024-12-16 by yorikvanhavre

kpemartin, 2024-12-16:
I'll try to stay afloat. Or maybe find a ladder to climb...

The self-contained Python macro would be a great help. I certainly want to get #17944 (or a replacement for it) alive again.

Andrei-Pozolotin, 2024-12-16:
@kpemartin :

in the spirit of floating and also climbing,  here is a crash pack: 

 [issue_18512.zip](https://github.com/user-attachments/files/18157226/issue_18512.zip)

contents:
``` sh
crash_18512.md # instructions
crash_18512.py # crash macro
crash_18512.FCStd # crash model
crash_18512.txt # crash stack report
```

the crash stack `crash_18512.txt` looks longer and different from the one posted above, but ends the same:
``` cpp
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x7867f424c1d0]
#1  0x7867f44c787e in std::_Rb_tree_decrement(std::_Rb_tree_node_base const*) from /usr/lib/libstdc++.so.6+0xe
#2  0x7867f69e8763 in Base::UniqueNameManager::PiecewiseSparseIntegerSet::Add(unsigned int) from /usr/lib/freecad/lib/libFreeCADBase.so+0x133
```

the `crash_18512.py` macro is a kitchen sink of relevant and irrelevant dependencies to reproduce original code closely