FreeCAD/FreeCAD #18513: Assembly: No "Create Simulation" command on MacOS FreeCAD 1.0.0

CompoundCAD, 2024-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When loading the example assembly in FreeCAD to test the motion capabilities the create simulation command is missing from the FreeCAD package for MacOS

### Full version info

```shell
FreeCAD_1.0.0-conda-macOS-arm64-py311.dmg
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-14:
@JohnOCFII could you confirm this ?

JohnOCFII, 2024-12-15:
I'm not familiar with the Assembly workbench, but I can confirm that the `Create Simulation` command is not in the Assembly menu in 1.0.0, but it **is** in the current weekly, 39642.

```
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.39642 (Git) Conda
Build type: Release
Branch: main
Hash: c80bb958f377e4d43c6a4075da173a79d153db68
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.6.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.33
  * Curves 0.6.52
```

[image]

[image]

luzpaz, 2024-12-15:
Thanks JohnOCFII.

hi @PaddleStroke Simulation will be in the next minor release, in your estimation ?

PaddleStroke, 2024-12-16:
This PR merged after 1.0 release. I don't think it will be backported to next point release. It's a new feature that needs more time for testing in dev branch. If you want to use it @CompoundCAD you'll have to use the weekly builds.
This should be closed

closed 2024-12-16 by luzpaz