FreeCAD/FreeCAD #18516: Zoom in causes constraint labels to become smaller instead of larger

rogerdahl, 2024-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When zooming in or out on a sketch, the text size of the constraint values becomes smaller when it should become larger, and vice versa.

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Anything else?

Screen recording:

https://github.com/user-attachments/assets/4d801fe6-c89f-4cc1-a9e3-f37ce0c13475

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-15:
I cannot reproduce on Windows
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.3
  * CurvedShapes 1.0.13
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.1
  * Ship 2024.11.26
  * Silk 0.1.5
```

Could you try this with a weekly build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

tarman3, 2024-12-15:
Can`t reproduce

<details>
  <summary>FreeCAD 1.1.0dev.39560 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.1; Silk 0.1.5; Curves 0.6.51

```
</details>

FlachyJoe, 2024-12-15:
I can reproduce in **perspective** view mode:

```
OS: Debian GNU/Linux trixie/sid (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39649 +1 (Git)
Build type: Debug
Branch: dev
Hash: 67b3fd3ecfa178eac7fb477c0d3ce545977ff027
Python 3.12.8, Qt 5.15.15, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.2.dev
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

3x380V, 2024-12-16:
That perspective mode problem is there for ages. I was advised to turn it off. Also see #7082

luzpaz, 2025-06-02:
soft bump

maxwxyz, 2025-08-19:
@Rexbas maybe

closed 2025-08-19 by maxwxyz