FreeCAD/FreeCAD #18518: FreeCAD 1.0.0 Saving Work document issue

noste99, 2024-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When saving an active work document for the first time I cannot find back the directory being a dynamic link to my archive partition. I find in the list of directories all but the dynamic link directory.

This happens in the FreeCAD 1.0.0 version I installed via snap on my Ubuntu 24.04 machine. The directory I use for all my work is listed after typing an ls command as ARCHIVE -> /archive/stefan/ARCHIVE/ . This directory does not appear in he list of locations shown when using the Save As function.

### Full version info

```shell
OS: Ubuntu Core 22 (MATE/mate/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1202
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-15:
Is this due to how Snap 'sandboxes' ? 
CC @furgo16 can you reproduce ?

furgo16, 2024-12-15:
What's a dynamic link?

noste99, 2024-12-15:
> naar: FreeCAD ***@***.***>
> datum: zondag 15 december 2024 13:37 CET

Sorry I mean symbolic link. I symbolically link an ARCHIVE directory in my home environment to a directory in a partition of a mounted disk. In Freecad when saving my work I see my home environment (files and directories) but not the symbolic link. 

 
I can work and find the version 1.0.0 working very well. Just this small inconvenience. 
  
Best regards, and thanks in advance, 
Stefan 
  
  
 
 
 
 

-----Originele bericht-----

furgo16, 2024-12-15:
@noste99 got it, thanks. I think running this command on the terminal should sort it out:

```shell
sudo snap connect freecad:removable-media
```

(assuming your FreeCAD snap installation is named `freecad`). [This issue](https://github.com/realthunder/FreeCAD_assembly3/issues/1241) has more detailed info on the `removable-media` snap plug.

noste99, 2024-12-27:
Hi again, 

Sorry to bother, but your suggested solution does not solve my problem. My linked disk space is not a removable media. I looked in the doc you mentioned but did not found solutions. 

Thanks in advance, 

Best regards, 

Stefan 

Closed [ https://github.com/FreeCAD/FreeCAD/issues/18518 | #18518 ] as completed.

luzpaz, 2024-12-28:
@noste99 Please add your full [About](jhttps://wiki.freecad.org/About) info. Please don't respond via email but instead click on this link (#18518) and respond within Github to this ticket.

noste99, 2024-12-30:
Hi, unfortunately did your suggested actions not solve my problem. I scanned through the documentation you pointed to and saw that snap by design is containing the package in a way that writing in other partitions is not possible. I was thinking whether I could not put in place a symbolic link making the work-files of FreeCAD stored in a directory that is accessible to the snap package and visible in my ARCHIVE. I am not sure whether backups of my ARCHIVE will also take up the files in the symbolic linked directories.

luzpaz, 2024-12-31:
This ticket needs to be transferred to FreeCAD Snap repo.