FreeCAD/FreeCAD #18523: Arch example file opens with errors

mnesarco, 2024-12-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In a fresh install of main branch, opening the first example file shows a lot of Draft errors

[image]

```bash
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:04:47  pyException: Traceback (most recent call last):
  File "/FreeCAD/FreeCAD_nix/build/Mod/Draft/draftviewproviders/view_wire.py", line 142, in onChanged
    super(ViewProviderWire, self).onChanged(vobj, prop)
  File "/FreeCAD/FreeCAD_nix/build/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:04:47  pyException: Traceback (most recent call last):
  File "/FreeCAD/FreeCAD_nix/build/Mod/Draft/draftviewproviders/view_wire.py", line 142, in onChanged
    super(ViewProviderWire, self).onChanged(vobj, prop)
  File "/FreeCAD/FreeCAD_nix/build/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:04:47  pyException: Traceback (most recent call last):
  File "/FreeCAD/FreeCAD_nix/build/Mod/Draft/draftviewproviders/view_wire.py", line 142, in onChanged
    super(ViewProviderWire, self).onChanged(vobj, prop)
  File "/FreeCAD/FreeCAD_nix/build/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:04:47  pyException: Traceback (most recent call last):
  File "/FreeCAD/FreeCAD_nix/build/Mod/Draft/draftviewproviders/view_wire.py", line 142, in onChanged
    super(ViewProviderWire, self).onChanged(vobj, prop)
  File "/FreeCAD/FreeCAD_nix/build/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:04:47  pyException: Traceback (most recent call last):
  File "/FreeCAD/FreeCAD_nix/build/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'

``` 

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39588 (Git)
Build type: Release
Branch: main
Hash: ad419828b00c6a11c47fb486de00aacaa1a96702
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-15:
Confirmed.

<details><summary>Expand for Report Output</summary> 

```shell
20:19:21  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAOKbpBo/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:19:21  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAOKbpBo/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:19:21  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAOKbpBo/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:19:21  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAOKbpBo/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:19:21  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAOKbpBo/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:19:21  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAOKbpBo/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:19:21  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAOKbpBo/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:19:21  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAOKbpBo/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:19:21  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAOKbpBo/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:19:21  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAOKbpBo/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:19:21  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAOKbpBo/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:19:21  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAOKbpBo/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:19:21  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAOKbpBo/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
[... 1174 lines cut ...]
    super(ViewProviderWire, self).onChanged(vobj, prop)
  File "/tmp/.mount_FreeCAOKbpBo/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:19:21  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAOKbpBo/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
 ```

</details>

```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39642 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: c80bb958f377e4d43c6a4075da173a79d153db68
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

Roy-043, 2024-12-15:
Caused by #18298.

closed 2024-12-17 by Roy-043

mnesarco, 2024-12-17:
BIMExample still opens with errors

[image]

<details>
<summary>Errors</summary>

```
17:25:39  pyException: Traceback (most recent call last):
  File "/FreeCAD/FreeCAD_nix/build/Mod/BIM/ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "/FreeCAD/FreeCAD_nix/build/Mod/BIM/ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
       ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/d3x1bxzq2750na7ccarh5sm7qw1ibv6h-python3-3.12.7-env/lib/python3.12/site-packages/pivy/coin.py", line 3560, in __getattr__
    raise e
  File "/nix/store/d3x1bxzq2750na7ccarh5sm7qw1ibv6h-python3-3.12.7-env/lib/python3.12/site-packages/pivy/coin.py", line 3552, in __getattr__
    return SoBase.__getattribute__(self, name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
17:25:39  pyException: Traceback (most recent call last):
  File "/FreeCAD/FreeCAD_nix/build/Mod/BIM/ArchSite.py", line 948, in updateData
    self.onChanged(obj.ViewObject,"SolarDiagramPosition")
  File "/FreeCAD/FreeCAD_nix/build/Mod/BIM/ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "/FreeCAD/FreeCAD_nix/build/Mod/BIM/ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
       ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/d3x1bxzq2750na7ccarh5sm7qw1ibv6h-python3-3.12.7-env/lib/python3.12/site-packages/pivy/coin.py", line 3560, in __getattr__
    raise e
  File "/nix/store/d3x1bxzq2750na7ccarh5sm7qw1ibv6h-python3-3.12.7-env/lib/python3.12/site-packages/pivy/coin.py", line 3552, in __getattr__
    return SoBase.__getattribute__(self, name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
17:25:39  pyException: Traceback (most recent call last):
  File "/FreeCAD/FreeCAD_nix/build/Mod/BIM/ArchSite.py", line 946, in updateData
    self.onChanged(obj.ViewObject,"SolarDiagram")
  File "/FreeCAD/FreeCAD_nix/build/Mod/BIM/ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "/FreeCAD/FreeCAD_nix/build/Mod/BIM/ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
       ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/d3x1bxzq2750na7ccarh5sm7qw1ibv6h-python3-3.12.7-env/lib/python3.12/site-packages/pivy/coin.py", line 3560, in __getattr__
    raise e
  File "/nix/store/d3x1bxzq2750na7ccarh5sm7qw1ibv6h-python3-3.12.7-env/lib/python3.12/site-packages/pivy/coin.py", line 3552, in __getattr__
    return SoBase.__getattribute__(self, name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
17:25:39  pyException: Traceback (most recent call last):
  File "/FreeCAD/FreeCAD_nix/build/Mod/BIM/ArchSite.py", line 946, in updateData
    self.onChanged(obj.ViewObject,"SolarDiagram")
  File "/FreeCAD/FreeCAD_nix/build/Mod/BIM/ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "/FreeCAD/FreeCAD_nix/build/Mod/BIM/ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
       ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/d3x1bxzq2750na7ccarh5sm7qw1ibv6h-python3-3.12.7-env/lib/python3.12/site-packages/pivy/coin.py", line 3560, in __getattr__
    raise e
  File "/nix/store/d3x1bxzq2750na7ccarh5sm7qw1ibv6h-python3-3.12.7-env/lib/python3.12/site-packages/pivy/coin.py", line 3552, in __getattr__
    return SoBase.__getattribute__(self, name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
17:25:39  pyException: Traceback (most recent call last):
  File "/FreeCAD/FreeCAD_nix/build/Mod/BIM/ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "/FreeCAD/FreeCAD_nix/build/Mod/BIM/ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
       ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/d3x1bxzq2750na7ccarh5sm7qw1ibv6h-python3-3.12.7-env/lib/python3.12/site-packages/pivy/coin.py", line 3560, in __getattr__
    raise e
[... 201 lines cut ...]
  File "/FreeCAD/FreeCAD_nix/build/Mod/BIM/ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
       ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/d3x1bxzq2750na7ccarh5sm7qw1ibv6h-python3-3.12.7-env/lib/python3.12/site-packages/pivy/coin.py", line 3560, in __getattr__
    raise e
  File "/nix/store/d3x1bxzq2750na7ccarh5sm7qw1ibv6h-python3-3.12.7-env/lib/python3.12/site-packages/pivy/coin.py", line 3552, in __getattr__
    return SoBase.__getattribute__(self, name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
```

</details>

```
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39677 (Git)
Build type: Release
Branch: main
Hash: b64549b348824653ef7880203ec6cefe031c5d1f
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/

```

yorikvanhavre, 2024-12-18:
That error will be fixed with #18578

PR #18558 (2024-12-17): Draft: Fix handling of coin nodes
Fixes #18523.

`find_coin_node` code taken from #18429.
commit db12b31b (2024-12-17): Draft: Fix handling of coin nodes

Fixes #18523.