FreeCAD/FreeCAD #18524: Crash on export to VRML

HuibSmeets, 2024-12-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I export below simple model (of a heatsink) to VRML FreeCad crashes. 
Export to STEP or WEBGL goes without issues.

[image]

### Full version info

```shell
OS: macOS 15.1.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Render 2024.8.4
  * A2plus 0.4.67
  * fasteners 0.5.28
  * lattice2 1.0.0
  * sheetmetal 0.4.25
  * Curves 0.6.47
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-15:
Can you provide the example file (upload as .zip)?

HuibSmeets, 2024-12-15:
This is the one:

[SK 431-1-25.FCStd.zip](https://github.com/user-attachments/files/18140293/SK.431-1-25.FCStd.zip)

wwmayer, 2024-12-17:
No problem with:
```
OS: Ubuntu 22.04.5 LTS (XFCE/xubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git)
Build type: debug
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/Classic theme/Qt default
```

Do you get any details about the crash from the OS?

HuibSmeets, 2024-12-17:
A .txt file with the MacOS crashreport attached.

[MacOS Crash Report.txt](https://github.com/user-attachments/files/18171161/MacOS.Crash.Report.txt)

wwmayer, 2024-12-17:
Thanks.

Looks like a problem with Coin:
```
Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libCoin.80.0.3.dylib          	       0x107ac1eac SoCallbackAction::~SoCallbackAction() + 68
1   libCoin.80.0.3.dylib          	       0x107adeea0 SoToVRML2ActionP::~SoToVRML2ActionP() + 260
2   libCoin.80.0.3.dylib          	       0x107add56c SoToVRML2Action::~SoToVRML2Action() + 40
3   libFreeCADGui.dylib           	       0x105317298 Gui::SoFCDB::writeToVRML(SoNode*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>&) + 236
4   libFreeCADGui.dylib           	       0x105317370 Gui::SoFCDB::writeToVRML(SoNode*, char const*, bool) + 60
5   libFreeCADGui.dylib           	       0x105319694 Gui::SoFCDB::writeToFile(SoNode*, char const*, bool) + 136
6   libFreeCADGui.dylib           	       0x104fca0ac Gui::Application::sExport(_object*, _object*) + 1132
```

HuibSmeets, 2025-01-09:
FYI: It crashes also on the older MacOS Monterey (12.7.6) for Intel Architecture.

```
OS: macOS 12.7.6
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * A2plus 0.4.60h
  * kicadStepUpMod 10.17.1
  * fasteners 0.4.54
  * ExplodedAssembly
  * Assembly3 0.11.4
```

MacOS Crashreport:
[crashreport Intel Mac.zip](https://github.com/user-attachments/files/18364280/crashreport.Intel.Mac.zip)

m1cha1, 2026-09-14:
No problem with:
```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.14.6, Qt 6.11.1, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.6.2, boost 1_91, Eigen3 5.0.1, PySide 6.11.1
shiboken 6.11.1, SMESH 7.7.1.0, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.0.p1-4f95ecaa
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.414/1.25
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 26.10.28.01.260707/ATI Technologies Inc./AMD Radeon(TM) 890M Graphics
Installed mods:
  * Addon Manager 2026.8.18
  * Curved Shapes 1.0.15
  * Curves workbench 0.6.81
  * Fasteners Workbench 0.5.64
  * SheetMetal Workbench 0.8.22
```

luzpaz, 2026-09-22:
@johnocfii can you kindly test on macOS dev build ? Thanks!

JohnOCFII, 2026-09-23:
> [@JohnOCFII](https://github.com/JohnOCFII) can you kindly test on macOS dev build ? Thanks!

No crash, but I did notice that the exporter does not change the extension from what ever file type you previously have selected.  If you manually change the file type, it seems to export correctly.

Here is the zipped file: 

[SK 431-1-25-Body.wrl.zip](https://github.com/user-attachments/files/32544599/SK.431-1-25-Body.wrl.zip)

[image]

```
OS: macOS 27.0
Architecture: arm64
Version: 26.3.0dev.48865 (Git)
Build date: 2026/09/22 20:26:30
Build type: Release
Branch: main
Hash: 019f5c50a945743b89cabf422ac5f5f1e297d890
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
QPA/File dialog/Color dialog: cocoa/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
OpenGL version/vendor/renderer: 2.1 Metal - 91.7/Apple/Apple M5
Installed mods:
  * OpenTheme 2025.5.20
  * Telemetry 1.0.7
```

flaviut, 2026-11-02:
Seems to be related to the specific compiler (clang vs. gcc) where Clang optimizes some undefined behavior:

|Coin | Compiler | Result |
|---|---|---|
| 4.0.3 (gcc, Nix) | GCC 15 | survives (guard kept) |
| 4.0.3 (clang, -O3) | Clang 21 | SIGSEGV, exit 139 |
| 4.0.10 (clang, -O3) | Clang 21 | survives (explicit guard) |
| 4.0.10 bundled (gcc) | GCC 15 | survives |

It's not a mac-specific issue, except that the default compiler on Mac is Clang.

This was fixed by https://github.com/coin3d/coin/commit/938bcc7c7f0e6098929c5dfacea222e5a2e7473d in Coin v4.0.4.

flaviut, 2026-11-02:
Fixed in FreeCAD 26.3rc1

closed 2026-11-02 by flaviut