FreeCAD/FreeCAD #18527: Assembly: selecting a joint in the tree should highlight the joined elements

eblanshey, 2024-12-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It is currently difficult to see which elements are joined together in an assembly. Toggling the joint currently displays two "origin" symbols, but in assemblies with many small parts, it's not enough to see which parts are being joined.

It would be great if selecting the joint could cause the elements I originally selected to be highlighted. For example, a revolute joint would highlight the circular edges and/or cylindrical faces. This would clearly allow me to see which parts have been joined, and how.

The highlighting feature could be triggered in the following ways (not all are needed):

- Selecting a joint in the feature tree
- Toggling the visibility of the joint in the feature tree
- Double clicking the joint to edit it
- In the Create Joint dialog, clicking one of the elements in the list (perhaps this could un-highlight the non-selected element)

Ideally the features would be highlighted even if obstructed by other objects, with some kind of transparency view. But that's a nice-to-have and not as important.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * WebTools 1.0.0
  * Help 1.0.3
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mgth, 2025-08-22:
A would propose a command and/or context-menu action “Select joints by selection” that selects (or filters the tree to) all joints that reference the current selection (object or sub-element). Optionally highlight them in 3D and allow quick visibility toggle.
Another approach would be to add a node for each object in an assembly, containing the ‘joints’ associated with that object; that’s what SOLIDWORKS does, for example.

PaddleStroke, 2025-08-22:
Sound good. We could use something similar to the preview shape to show the selected entities. I'll dig that later

PaddleStroke, 2025-08-28:
https://github.com/user-attachments/assets/e4ec36af-70c6-4449-937e-104ba6c83ffc

PaddleStroke, 2025-08-29:
So this is facing the exact same issue as #17818 which is making it very difficult to solve.

I need to investigate since those are important quality of life features that we need in assembly...

PaddleStroke, 2025-10-23:
This feature is surprisingly extremely hard to make. You'd think there's an easy way to highlight something but no it's kind of a mess.
The good news is that I think I'm getting closer to a solution.

@eblanshey point color will be overriden but as you mentioned they are not very noticeable. In particular since often the JCS is drawn on top of it. Unless you have some offset from that point. In which case it can be seen.

closed 2026-01-22 by kadet1090