FreeCAD/FreeCAD #18531: Problem with parameters and dimension lines

lepesite, 2024-12-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When parameter values ​​are entered, they are copied to the Expression editor just fine, but they almost never go to, for example, the Insert length dialog.
The parameter value can be used as part of a parameter value calculation formula and it works OK.
Occasionally, the parameter value is linked to the dimension line value and it works perfectly.
Another disadvantage is that even though the value has not been transferred to the dimension line value, the line color changes to brown, indicating that the dimension line value is based on the linked parameter.

I have not found a solution to this problem!

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Finnish/Finland (fi_FI)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * AirPlaneDesign 0.4.1
  * AnimationFreeCAD 1.0.0
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * Help 1.0.3
  * Manipulator 1.5.7
  * Movie 2024.2.15
  * OSE3dPrinter
  * pcb 6.2023.1
  * QuickMeasure 2022.10.28
  * Render 2024.12.14
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-15:
What is the actual issue here? Do you want to see the expression in the length dialog?

luzpaz, 2024-12-18:
It sounds like the request is just that, for the expression to conveniently be mirrored in the length dialog in order to not have to add the extra step to copy/paste it into the other dialog.

lepesite, 2024-12-18:
Thanks luzpaz for the very quick response. I was following the YouTube "FREECAD 1.0 Release - Ultimate Complete Beginner Tutorial" starting at 7:00. The instructions and mouse cursor movements made it so that I didn't notice that the default Type setting was something other than "Length". I didn't see any reference to this option in the instructions. When I noticed it and changed [ Type > App:PropertyLength,  ] everything went just fine. Lots of compliments from the support.Pepe 😊Sent from my mobile phone-------- Original message --------From: luzpaz ***@***.***>Date: Wed, 18 Dec 2024, 14.21To: FreeCAD/FreeCAD ***@***.***>Cc: lepesite ***@***.***>, Mention ***@***.***>Subject: Re: [FreeCAD/FreeCAD] Problem with parameters and dimension lines (Issue #18531)
It sounds like the request is just that, for the expression to conveniently be mirrored in the length dialog in order to not have to add the extra step to copy/paste it into the other dialog?
ping @lepesite

FEA-eng, 2025-09-17:
Isn't it more of a feature request than a bug then?

luzpaz, 2025-09-18:
Any proposal for the rename of this feature request to be more succinct ?

luzpaz, 2026-09-23:
Updated title of ticket.