FreeCAD/FreeCAD #18534: Bug in Pocket Operation: Start At doesn't work in some circumstances.

Connor9220, 2024-12-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Under Start At under properties, you have option of doing Center, or Edge.  Looks like it doesn't matter what this is set to, , if you have minimum travel option set to TRUE it starts at the edge.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * ThreadProfile.backup1725502069.7449253 1.90.0 (Disabled)
  * QuickMeasure 2022.10.28
  * dodo 1.0.1
  * btl.backup1725502066.664948 0.9.9 (Disabled)
  * btl.backup1720304573.9159045 0.9.9 (Disabled)
  * OpenTheme 2024.9.1
  * A2plus.backup1725502065.0763218 0.4.65 (Disabled)
  * btl 0.9.9 (Disabled)
  * OpenDark 2024.3.13
  * A2plus 0.4.68
  * btl.backup1717184435.3801382 0.9.9 (Disabled)
  * lattice2 1.0.0
  * ThreadProfile 1.92.0
  * fasteners.backup1725502068.289526 0.5.24 (Disabled)
  * fasteners 0.5.31
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-16:
Can you provide an example file and test with the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Connor9220, 2024-12-16:
Here is a screen shot of the options.  Attached is the test file.  The operation in question is POCKET.  You'll want to run a sim of it as is, and you'll see that it starts on the outside edge.  If you change the Min Travel to false, recompute and re-run the sim, you'll see that it starts in the middle like it's suppose too.

[image]
[slot_test.zip](https://github.com/user-attachments/files/18155405/slot_test.zip)

luzpaz, 2024-12-16:
@shaise do you have a moment to test this ?

Connor9220, 2025-01-08:
Another interesting thing.  If I have multiple pockets in the same op, with Min Travel set to False, it always goes to X0 Y0 after each pocket.  When set to True, it doesn't.  I'm not sure why it would ever go to X0 Y0 after a feature.. it's a wasted move.

FEA-eng, 2025-09-17:
@tarman3 Do you know if it's still relevant?

tarman3, 2026-01-11:
Probably can be fixed by 
- #25313

closed 2026-09-28 by Connor9220