FreeCAD/FreeCAD #18535: Tool dock doesn't show.

Connor9220, 2024-12-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The tool dock doesn't show up.  You have to click the pencil icon to get it to show up after you bring the tool doc back up.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * ThreadProfile.backup1725502069.7449253 1.90.0 (Disabled)
  * QuickMeasure 2022.10.28
  * dodo 1.0.1
  * btl.backup1725502066.664948 0.9.9 (Disabled)
  * btl.backup1720304573.9159045 0.9.9 (Disabled)
  * OpenTheme 2024.9.1
  * A2plus.backup1725502065.0763218 0.4.65 (Disabled)
  * btl 0.9.9 (Disabled)
  * OpenDark 2024.3.13
  * A2plus 0.4.68
  * btl.backup1717184435.3801382 0.9.9 (Disabled)
  * lattice2 1.0.0
  * ThreadProfile 1.92.0
  * fasteners.backup1725502068.289526 0.5.24 (Disabled)
  * fasteners 0.5.31
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-16:
Can you please provide a screenshot of what is missing?
Please test if this happens on the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

JohnOCFII, 2024-12-18:
I've seen this many times as well.  Sadly, it is random.

Connor9220, 2025-01-19:
> I've seen this many times as well. Sadly, it is random.

It's not random.  It's on initial load.  Once it's been opened, if you start a new job, or close and re-open the dock, it works.  Restart FreeCAD, and open it, tools are missing.  You have to click the edit button and then close the dialog for them to show up.

closed 2025-02-03 by chennes