FreeCAD/FreeCAD #18536: C++ error

PRobwar, 2024-12-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Im getting this error message:
21:54:33  C++ exception thrown (Unable to find zip structure: End-of-central-directory)
21:54:33  C++ exception thrown (Unable to find zip structure: End-of-central-directory) 
every second  

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Dark/Qt default
Installed mods: 
  * Alternate_OpenSCAD 1.0.0
  * FeedsAndSpeeds 0.5.0
  * freecad.gears 1.0.0
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Anything else?

was there from first time start up

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-16:
Is this related to the zipios issue? (#18494)

wwmayer, 2024-12-17:
Do you have the Start page open? Do the errors go away if you close it?

PRobwar, 2024-12-18:
As far as I know I did not have the start page open at the time will have a
look when I get home later today.

wwmayer, 2024-12-18:
There are two classes that throw exceptions with this error text: ZipHeader and ZipFile.
ZipHeader currently is only used when opening 3MF meshes which probably isn't the case here. And ZipFile is used for 3MF meshes and for generating the thumbnails that are shown on the start page.

kriewaldt, 2024-12-19:
I had the error occur at the same time.
I loaded the latest build with the same error.
I then renamed the user.cfg and system.cfg files and reloaded.
Fixed the problem. Hope this is helpful. I don't know what caused the problem.
It was working until I opened a new session.

PRobwar, 2024-12-20:
Hi
I started with the start screen which I did not have setup when I update to latest version. Now there error I had has gone.

Paul Warwick

wwmayer, 2024-12-20:
I guess that in your recent file list there must have been a corrupted FCStd file that ZipFile isn't able to open. After renaming/deleting the user.cfg you start with an empty recent file list and that's why it fixes the problem.

melody8801, 2025-01-20:
in my case, the renaming (deleting) of the Files usr.cfg and system.cfg was not enought. 

I had 1.0.0RC2 installed. Maybee there was an issue with an installed mod under the folder %appdata%/freecad/mod. 
I just renamed the whole folder %appdata%/freecad to %appdata%/frecad.bak. That solved the isssue in my case. 

i had installed the following mods:
- A2plus
- Assembly4
- fasteners
- freecad.gears
- kicadStepUpMod
- Render

the Plugins came from my previus installation of FreeCAD 0.21

schwukas, 2025-01-21:
In my case the issue was that FreeCAD tried to load a recent file that was apparently having issues. In `.config/FreeCAD/user.cfg` I've removed the problematic entry from `<FCParamGroup Name="RecentFiles">`. Trial and error or just clearing that whole list might resolve the problem.

TolloJenkins, 2025-05-15:
у меня появилась таже самая проблема.( C++ exception thrown (Unable to find zip structure: End-of-central-directory) )
и решилось удалением файлов usr.cfg и system.cfg. они находятся по пути:
C:\Users\<имя пользователя>\AppData\Roaming\FreeCAD

перестановка не помогла видимо из-за того, что я не удалял пользовательские данные.

luzpaz, 2025-05-15:
@TolloJenkins please edit and add english translations, thank you.

FEA-eng, 2025-09-27:
@PRobwar or anyone else participating - is it still relevant?

ElementW, 2026-04-13:
Managed to reproduce on 1.0.2 with with [`blank_zip_central_directory_missing`](https://github.com/ElementW/FreeCAD-evil-files/blob/c2630035b75ec0eeb3f73823cba3c5a32e3ea730/blank_zip_central_directory_missing.FCStd), [`blank_zip_central_directory_missing_noPK`](https://github.com/ElementW/FreeCAD-evil-files/blob/c2630035b75ec0eeb3f73823cba3c5a32e3ea730/blank_zip_central_directory_missing_noPK.FCStd) and [`blank_zip_central_directory_truncated`](https://github.com/ElementW/FreeCAD-evil-files/blob/c2630035b75ec0eeb3f73823cba3c5a32e3ea730/blank_zip_central_directory_truncated.FCStd) from my [evil files repo](https://github.com/ElementW/FreeCAD-evil-files), by injecting them into the MRU list in `user.cfg`.

Couldn't reproduce in 1.1 onwards however.

<details>
<summary>1.0.2 version info</summary>

```
OS: Void Linux (KDE/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

</details>